eh2k/squares-and-circles #3: CD > Enveloppe : trig not trig

PooLP, 2022-04-16:
Hi,
Apparently, the new version have a bug, the enveloppe not work.
i'm put my GATE output from CV/Gate interface to TR1 or other TR input of O&C, but the enveloppe is not triggered.

I do not have the problem with DRUMS or M-OSC part, all TR input work, and I/O test all trig work.

Another question, Is it planned to integrate a part of native O&C and Hemisphere modules (sequencer, Qantizer ...) ?

Thanks for great work ;)

ps : The problem appear with old version aslo.

eh2k, 2022-04-17:
Hi - @PooLP, 
thanks for reporting the bug. It does not work for me either - will try to fix it at the next opportunity. 

I think I will not try to port the hemisphere apps directly. But it might be possible - if I find the time/motivation - that a sequencer in combination with randomness will follow in the future. A quantizer would certainly not be very complex - as far as I know hemisphere uses the braids code for that.

eh2k, 2022-04-24:
the bug is fixed in a24c2ce3cb2c9c947c12f62303724ffb03d4f27c
Check the latest firmware (https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1107901164)

closed 2022-04-24 by eh2k

PooLP, 2022-05-28:
Sorry for delay, it's works 👍