eh2k/squares-and-circles #5: compiling for USB midi

oldmanfury, 2022-04-24:
If possible, can you try to compile a hex file to enable usb-midi?  (I tried, ran into some errors, and realized this is way beyond me at this point).  

Motivation: I made a USB breakout-panel using a Synthrotek power-PAK.  Being able to use square-and-circles with the monome-teletype via I2C2MIDI would be phenomenally powerful, and would clean-up a bunch of patch-cables.

eh2k, 2022-04-26:
Hello @oldmanfury,

I have played around a bit with usb_midi. Since I don't have any other usb-host devices besides my laptop, I tested it a bit with VCV. Works actually so far quite well. 

You can have a look at the code in [midi.cxx](https://github.com/eh2k/squares-and-circles/blob/dev/src/midi.cxx) in the dev branch.
I have redesigned the MidiHandler a bit - if I don't find any other problems, the changes will be transferred to the main branch at the next opportunity. I'm not sure yet if I want to support usb_midi in general in the future.

Here are a few more hints:
  * You have to compile with `-DUSB_MIDI_SERIAL`
  *[midi.cxx:215](https://github.com/eh2k/squares-and-circles/blob/dev/src/midi.cxx#L215) - The loop function is called cyclically by the timer. Within the loop function the incoming midi bytes from `usb_midi_read_message` are passed into the MidiHandler.

oldmanfury, 2022-04-27:
Works like a charm!  Having the USB midi enabled is so powerful, I'm able to write scripts on the monome teletype addressing all four engines.  Teletype i2c2midi module is a host, so it is a direct-connection.  Very cool!

oldmanfury, 2022-04-30:
Having much fun with this.  Found a bug, it seems.  Set up squares and circles four engines for midi ch1,2,3,4.  Play notes via a sequencer to each of those channels, and CH1 doesn't respond.  I can't get all four channels to respond at once - in fact if I chose arbitrary channels like 5,6,7,8 it kind of flips out (hung note loud noises).  Could you add MIDI CC's as modulation sources at some point?

eh2k, 2022-05-03:
I suspect that somehow there is not a follow NoteOff message for every NoteOn. As a result, > 4 incoming NoteOns are ignored. See NoteOn/NoteOff methods in the midi.cxx.

oldmanfury, 2022-05-05:
I tried using midi-ox since it has an event window I can really monitor what I've sent the SQ&CIRC.  It is behaving really oddly with different midi channels on each engine.  Just cannot get all four to respond - usually it is output A that is completely incommunicado.  The problem isn't just missing note-offs (hanging notes).

jaxoncute, 2022-05-12:
@oldmanfury  - any chance you can post the .hex to load with Teensy? I'm no programmer at all but big Elektron fan and this firmware looks awesome! 

Also, any tips on which OC firmware has the most FX options? I love hemispheres for the dual functionality but miss my super disting sometimes for the dual FX options.  Thank you

eh2k, 2022-05-13:
@jaxoncute 
I assume you can't connect the devices directly via USB, or can the new Elektron devices function as USB hosts? I assume you are also using a computer/over hub here?

eh2k, 2022-05-13:
> I tried using midi-ox since it has an event window I can really monitor what I've sent the SQ&CIRC. It is behaving really oddly with different midi channels on each engine. Just cannot get all four [...]

@oldmanfury I gues the issue is in the NoteOn,NoteOff,process methods in the midi.cxx. Maybe the /src/stmlib/algorithms/note_stack.h is the solution.

jaxoncute, 2022-05-13:
@eh2k I was just hoping to find a .hex so I could install this MachineDrum style firmware to O&C with ease. I'm not much of a techy at all so I'm used to using Teensy when swapping firmware and couldn't find a hex. It sounded like @oldmanfury had created one.

eh2k, 2022-05-13:
@jaxoncute - the firmware is still work-in-progress. You can find the latest firmware builds here https://github.com/eh2k/squares-and-circles/actions, any feedback is welcome.

eh2k, 2022-06-30:
Close - Midi-USB is available in RC_0.0k. Files of interest:
 * https://github.com/eh2k/squares-and-circles/blob/main/src/polyVA.cxx
 * https://github.com/eh2k/squares-and-circles/blob/main/src/midiMonitor.cxx
 * https://github.com/eh2k/squares-and-circles/blob/main/src/midiClock.cxx

closed 2022-06-30 by eh2k