eh2k/squares-and-circles #6: Output-routing / Mixing (discussion)

eh2k, 2022-04-30:
Regarding the output routing/mixing suggestions - I would use this issue as a starting point.

Here are the quotes from the general feedback:

> The drums are so good though. Wondering if it would be possible to have a level control for all the drum engines?
Also would it be possible have a mixing option to allow engines to choose an output? Id love to be able to set up a multi oscillator or a set of drums and have the come out of one output.

> A feature request (is this the right place for that?) - being able to send the 4 engines to any of the 4 outputs would be really handy. There are times where I'd like all four on one channel, just so I don't have to use a dedicated mixer module prior to routing the audio elsewhere. Just a 'output' row where you select A,B,C,D would be great. Maybe that would make the reverb modules more versatile too - just route the signals you want reverbed to C or D (if the reverb is using those outputs).

My reply was/is:
> Mixing option: This is really tricky - i don't have a simple solution for this - especially if you add stereo or panning etc..

Here some example setups - keep in mind that it is possible to put stereo engines on "even" outputs, so they will become mono if the neighboring "odd" output engine is not an fx. 

[image]

oldmanfury, 2022-04-30:
So much more complicated than I imagined!  I was thinking that you select an arbitrary destination for each engine (A,B,C,D).   If C has an effect on it, a synth engine pointing at C goes through the effect.  Hadn't thought about routing to both inserts and outputs - looks versatile.  What is AUX?  I think I saw something about hijacking CV1 to make an AUX input?

eh2k, 2022-05-01:
Yes, the AUX can be configured using one of the CV-Inputs (prefer CV-4 for better quality)

One more note to the diagrams:
 * The circled blocks are "adjustable parameters" within the Machine I/O Config.

Well, basically it's a simple logic:
 * Mono Engine: Output is straight routed to the DAC
 * Stereo engine: output is routed to DAC 1+2, DAC 2+4 depending on odd/even channels
   * Currently, all stereo engines process audio input - consequently, the signal of the "previous" engines can be inserted.
   * The directly neighboring engine is "100% wet" inserted by default, the output of the other previous engines can be added as "send" signal. 
   * If the insert signal of the directly neighboring engine is reduced, it will be processed as a dry/wet mix - the dry signal will then be mixed to the stereo fx signal.
   * Because of the processing order, it is not easy to use e.g. the signal of engine 4 as input for engine 1.

Hope the routing / mixing logic is now more understandable.

andersnils, 2022-05-08:
Can you clarify on how insert 3 works in the second example (3 mono + one stereo fx)? Does having it at the highest number/amount mean 100% goes into the fx engine and the dotted line bypass is removed?

Thanks, just installed this today and its great so far!

eh2k, 2022-05-09:
Hi

The dotted signals are "muted" by default. The insert-3 is set to 100% by default, so the signal is replaced by the stereo output of the following fx-engine. You can add the signals from the previous engines by increasing the other insterts. E.g. a snare on engine-2 has a dry output on DAC2 and a 20% stereo wet output on DAC3+4. If you want reduce the fx e.g. hihats on engine-3 to 30% by decreasing the insert-3 it would result in lower volume - so the other 70% are mixed to the output behind the FX (insert-3 = dry/wet mix), because there is no direct DAC output.

Currently the gain can only be positive - I think lowering the volume with gain is missing (specially in case of insert-3 dry signal - because you cannot do it by external mixer)

zhincision, 2023-05-14:
Hi. Can you please add simple tutorial, for example, how to use stereo reverb

eh2k, 2023-07-28:
This issue is already a while ago, but the current version has simplified a lot from my point of view. Overall, the concept is now general and can also be operated wonderfully on my Daisy-based prototype (two audio outputs or even up to eight).

In short:
  * Each engine has a level and pan parameter.
  * Each engine can be assigned to any mono output or to two for stereo. It is also possible to set no output (or muted), e.g. to feed in the engine completely from an FX engine.
  * All engines can be fed into FX engines, regardless of the slot order.

In total, the previous routing can be mapped in the same way. However, due to the panning, there are even more use cases. You can also put all four engines on one output. At the moment there is still the limitation to four engines, but let's see...

At moments, the default setting could be further optimised if necessary. For example, stereo engines are mapped to two outputs by default (previous behaviour). CV engines are treated similarly, although mixing with audio makes no sense here.

Further suggestions or proposals welcome.

closed 2024-01-14 by eh2k