eh2k/squares-and-circles #8: Calibration troubles

BAREBONESLDN, 2022-05-16:
Hi,

When I hold down the left button on power up I get the test menu but I can't seem to access the calibration menu?
Any help appreciated.
Thanks!

spiilmusic, 2022-05-16:
Hi @BAREBONESLDN 
It seems that you need to update to the new firmware i.e. https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/firmware-a24c2.zip . The calibration feature is relatively new

BAREBONESLDN, 2022-05-17:
Unfortunately there is no hex file for the new version and I don't know how to compile it.  :(

spiilmusic, 2022-05-18:
@BAREBONESLDN what version of s_c are you using?
The [archive above](https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/firmware-a24c2.zip) already has the hex file you need.

BAREBONESLDN, 2022-05-18:
Thanks I figured out how to compile it, it wasn't too hard. :)

closed 2022-05-19 by eh2k