eh2k/squares-and-circles #10: Support presets / patch managment...

eh2k, 2022-07-01:
> 2. Save a limited number of presets that recall entire machine settings (ideally to be able to switch between 4 track drum machine vs 4 voice poly with out manually reassigning midi channels etc each time. Perhaps holding both encoders could open some save/patch menu.

_Originally posted by @MidiSlave in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1169369335_

#### Consideration:
At the moment, the patch is stored in the EEProm memory.
The size is not sufficient for several presets. But it would be conceivable to store it with LittleFS, which is gone after a firmware update. But possibly solvable for the future with the WebFlasher.

TimoRozendal, 2024-01-08:
This would be an awesome addition, really useful for live use.

eh2k, 2024-01-10:
I could imagine that pressing both encoders would open a simple "menu" where you could, for example, select one of four slots to save or load the patch.

However, I also wonder what exactly the use cases are. Midi sysex via USB might also be possible. This means that a third party device could request or load the Systex via Midi.

TimoRozendal, 2024-01-10:
For me the use case would be when I prepare a larger live set and where I can load a new 'preset' for a new track/piece. Manual clicking is fine then, sysex over usb is a bit of a puzzle without a computer.

TimoRozendal, 2024-04-15:
Awesome! looking forward to testing it

closed 2024-12-01 by eh2k