eh2k/squares-and-circles #13: DAC8564

goerserr, 2022-07-20:
Hi, I have 2xO&C built with DAC8564. The only difference in original firmware is that you must set RST pin to LOW instead of HIGH. I can't find similar DAC setup in s&c. Where should I look?

eh2k, 2022-07-21:
Hi, that's good to know about the alternative dac.

It is not directly possible to change hardware settings in the code - but I think it should not be too complicated to add a setting or similar in the next release.

eh2k, 2022-09-04:
Hi @goerserr,
It is now possible to execute certain setup commands. Could you please confirm that it works with the DAC8564.
Please flash the latest firmware with [image] and execute the setup command.

[image]

closed 2022-11-19 by eh2k