eh2k/squares-and-circles #14: tuning offset on the synth machines

eh2k, 2022-07-24:
Is anyone else experience a tuning offset on the synth machines ?  Even with the quantizer on (to make sure it's not my input v/oct signal that's the problem) I am experiencing synth engines about a half semitone high.  

I've run the DAC calibration routines a couple of times, but no difference. Weird.

Second point - the new calibration routines you've written are tremendous - so much easier than the original O&C routines - nice work eh2k.

_Originally posted by @scuzzphut in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1192006147_

spiilmusic, 2022-07-24:
> (comment)

Hi there. I noticed something like that when I tested base waveforms like
sin and others. Without any v/oct CVit can up or down by semitone.

eh2k, 2022-07-24:
@scuzzphut 

will be fixed in the next RC 0.0l. The cause is/was: The DAC is running with 50kHz instead of 48kHz due to a float/int division bug.

scuzzphut, 2022-07-24:
great stuff, eh2k.

eh2k, 2022-07-24:
I did the following test:

Set Engine: M-OSC:Virt.Analog
 * Set note C4
 * CV to '-' (or quantizer to semitones)

Comparing frequency in VCV with other vcv-oscillators with [NYSTHI/HotTuna](https://library.vcvrack.com/NYSTHI/HotTuna). After changing the [kSampleRate](https://github.com/eh2k/squares-and-circles/blob/main/lib/plaits/dsp/dsp.h#L48) to 50kHz, there was virtually no more tuning offset. However, I have corrected the DAC frequency in general so this has an effect on all engines.

> Just swapped the Teensy around and tested.  All 3 O&C exhibit a tuning offset (to varying degrees) when using a quantized v/oct input signal.  I'm guessing this is down to ADC / DAC calibration and I'm going to suggest adding a coarse and fine tune to each chromatic engine.
> 
> _Originally posted by @scuzzphut ...

eh2k, 2022-07-25:
With RC_0.0l the DAC is updated with 48kHz - the synths should now be tuned correctly. 

It would be great someone could test on another device and confirm this here...

scuzzphut, 2022-07-25:
I'll test tomorrow and report back. Thanks for the quick update !

scuzzphut, 2022-07-26:
Hi eh2k.  I installed the RC_0.0I and it's working well. Tuning is spot on with all chromatic engines. 
The web installer also worked first time without a hitch. Great work again !

eh2k, 2022-07-27:
Hi @scuzzphut, thanks for the reply. It's crazy - the off-tune has always been in there - and I don't think I've really done anything melodic yet - now suddenly everything sounds right :-)

closed 2022-07-27 by eh2k