eh2k/squares-and-circles #15: Firmware Innstalation problem

Arnfris, 2022-08-05:
Hello. I have a problem with the firmware instalation. I have flashed the firmare 0,1 from browser. I can't seem to get past the first page in the setup meny. I am able to flip the display 180 degrees an change the brightness but i am not able to go to the next page. After a while the oled screen goes "black" Any idea what i have done wrong? (I have the large O_C. 14 hp).

eh2k, 2022-08-06:
Hi @Arnfris, is there some Version Info on the PCB? Normally, you should switch through the calibration pages by pressing the buttons - which somehow doesn't work on your device. If the screen goes black - it is probably the screensaver after 5 minutes - if you turn the encoder or press a button it should come back on.

Arnfris, 2022-08-06:
Hello eh2k. Thank you for fast response. It is pcb revision 2e. (Bought from pusherman). I realize this might also be a hardware problem. (With my soldering). Is it the 2 smal black push buttons you use to scroll trough the setup pages? If that is the case i have a problem with one of my solder joints. (I am able to use one of the push buttons to see your logo and firmware revision, but the other one does not do any thing).

Arnfris, 2022-08-06:
Hello again. As this probably is a hardware problem I will do some more research on my own maby ask on the pusherman noob forum. Your energy is better used on firmware development :)

eh2k, 2022-08-06:
All right, then I'll close this issue.
If the "right" button does not work, you could theoretically use the other button after "flip180" to step through the calibration.

closed 2022-08-06 by eh2k