eh2k/squares-and-circles #17: Module does not power on after flashing

Arnfris, 2022-09-06:
Hello again. The latest firmware update 0.0m does not work on my build. I am using windows 10, the chronium browser and the web flasher. I have the 10 hp pcb revision 2e, dac85651. The problem occurs after flashing when the module (oled) does not "come to life". (The oled stays dark). Firmware 0.0l works fine and i am wery happy with it :) 
Regards 
Arnfris

themoonscientistandco, 2022-09-06:
Hallo ,i have the same problem as Arnfris ,I use a Mac ,version 0.0i word also fine
grtz
the moonsclientist

eh2k, 2022-09-06:
Helle @Arnfris, @themoonscientistandco 

thank you for the response.

I have just flashed the 0.0l (0.0L) back, saved the patch and then flashed the 0.0m again (to test whether this is somehow due to the loading of the "patch/preset"). Everything with WebFlasher on Chrome - without any problems for me. (Sometimes I get the "NotAllowedError: Failed to write the report" error, but this is somehow a peculiarity of browser, which no longer comes the next time you try) 

I have generally optimized the display timing in 0.0m - I could imagine that there are different display hardware versions that behave differently. Could you please test or tell me if you can hear something while the display is off? Actually, the "patch" saved with 0.0l should load with 0.0m at start-up. 

It would also be good to know if there are others where everything works without problems.

Arnfris, 2022-09-06:
Hello eh2k. 
thank you for fast response.
I will do it tomorrow and come back to you. :)
Regards
Arnfris

Arnfris, 2022-09-07:
Hello again eh2k. 
I have now updatet firmware to 0.0m and I do not think i have sound. I can not be completely sure   as i did the "meny dive" with a dark oled. I remebered the route to a drum sound, in rhe right position or flipped 180, gave it a trigger but no sound. I noticed that the oled stayed on during the update up to it was finished when it turned black. I use the blue oled from pusherman. (I try to give as much information ad possible because i do not know what is relevant :) ). 
regards
Arnfris

n0ahg, 2022-09-07:
Thanks for the great firmware.

I am also experiencing the same issue with a blank oled screen when trying 0.0m but everything works fine in 0.0l.

I am using a pusherman uoc with blue oled.

eh2k, 2022-09-07:
Hello @Arnfris, @themoonscientistandco, @n0ahg

Thank you very much for the response and troubleshooting.

I have now tested the whole thing again from another notebook (windows instead of linux), upgrade has worked from 0.0l to 0.0m as expected ?!

To exclude the WebFlasher, could you please try to flash the firmware with the TeensyLoader:
* https://www.pjrc.com/teensy/loader.html
* https://ehx.spdns.org/squares-and-circles/flash/0.0m.zip

Many thanks in advance

themoonscientistandco, 2022-09-07:
Hallo eh2k 
i flashed the hex file in the link with de teensyLoader
same problem ,black screen
grtz
The Moonscientist

Arnfris, 2022-09-07:
@eh2k @themoonscientistandco @n0ahg 
Yes, same result for me when using the teensy loader and the hex file. (Black screen). If somebody else also could do the "blind test" on fitmware 0.0m and listen for sound that would be grate to rule out that i did a mistake in the proses and fed a trigger to a lfo or adsr or something that does not produce sound. :)

eh2k, 2022-09-07:
@Arnfris, @themoonscientistandco 

hmm - I'm still unsure whether the Teensy hangs at start-up or there is an issue with the display.

I have a test website here (https://eh2k.github.io/□●/test/)  that allows you to connect directly to the Teensy and simulate the UI events and see the displaybuffer. The whole thing works completely without the O_C - just connect the Teensy via USB. If this works like in the video at you, it must be a display issue and I have to adjust the firmware or reverse my changes step by step.

I have tested this case with three different Teensy boards.

I don't know if there are any differences with the Teensy 4.0 - e.g. the bootloader (https://forum.pjrc.com/threads/67232-Teensy-4-What-triggers-Halfkay-to-do-a-full-flash-erase).

https://user-images.githubusercontent.com/11786292/188955474-38a348c5-488d-4511-9db3-c881fbb61cf1.mp4

themoonscientistandco, 2022-09-07:
Hallo. I did the "blind test" and I have sound,a nice clear 808 sound on the corresponding outputs t1 on out 1 and so on

Arnfris, 2022-09-07:
@eh2k @themoonscientistandco 
Ok. Then it seems like it is the display. Should i go on and conect using the above site? I tried but was not able to conect. The teensy loader did not recognize the teensy, maby becsuse i have cut the power trace on the back of the teensy.

eh2k, 2022-09-07:
@themoonscientistandco @Arnfris 

Ok - then it really seems to be the display. I'll try to make some adjustments in the firmware, maybe a setting ...

@Arnfris  - oh that's right, the teensy can also remain in the OC if the power connection is cut.

Thanks again for the feedback

eh2k, 2022-09-08:
Hello @Arnfris, @themoonscientistandco,

I have reversed two changes here:
 * Teensy platform 4.15.0 to 4.17.0 upgrade
 * Typo in the SPI Initialization - clock 4000000 instead of 400000.

Please try to flash the following hex again - and give me feedback if it works.

https://www.pjrc.com/teensy/loader.html
https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/firmware_00M.hex

Many thanks in advance

n0ahg, 2022-09-08:
Hi,

firmware_00M.hex
<https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/firmware_00M.hex>
doesn't
fix the black screen issue for me.  Flashing back to 0.0l and everything
works again.

Thanks
Ray

themoonscientistandco, 2022-09-08:
Hallo eh2K, Progresion , version 0.0M works perfect .
Many thx
Grtz
The moonsclientist

themoonscientistandco, 2022-09-08:
Hallo eh2K
I spoke too fast 
when i turned off the power of the modular and turned it back on, the screen was black again
after switching on and off several times, the screen worked again  
until I turned off the power again
the screen was black again
and so on

Arnfris, 2022-09-08:
@eh2k @themoonscientistandco @n0ahg 
Hello.
For me it turned black when i pressed the reboot on the teensy loader. It stayed on during the programing. (Had version 0.0l programed).

eh2k, 2022-09-08:
Hello @n0ahg @themoonscientistandco @Arnfris - thanks again for testing.

Trial no. 2: 
  * https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/firmware_00M2.hex

I have undone the ADC DMA MUX optimization - the process until the start screen disappears should be identical to 0.0l now. 

@themoonscientistandco - I just did the power on/off 10 times - the start screen was always there. If the screen remains black from the beginning, it should actually be due to the initialization.

@Arnfris - if you press the reboot or "reset in the WebFlasher" the teensy jumps into the bootloader, that is waiting for input. The buffer in the display remains until the next reboot.

n0ahg, 2022-09-08:
Hi,

Still a black screen on boot with 00M2.

Thanks
Ray

themoonscientistandco, 2022-09-08:
After starting the teensy 10,20 times with version 00m the screen problem besoms stable and the screen started up every time I pul up the power.  
I flashed version 00m2 , the problem was the same and came more stable after 10.20 teensy booted up like version 00m

Arnfris, 2022-09-08:
Hi,
Still black screen here too.
(I will be back tomorrow)
Regards
Arnfris

eh2k, 2022-09-08:
I'll try to test the behaviour tomorrow with another "diy" power supply, maybe I can reproduce it that way. Thanks for your help.

Arnfris, 2022-09-08:
@eh2k 
Ok, I try to do the flash tomorrow in my rack with my main power supply to see if that helps. (Right now i have used the supply i use outside the rack to test for shorts as it brings me closer to my laptop). 
regards
Arnfris

Arnfris, 2022-09-09:
@eh2k @themoonscientistandco @eh2k 
Hello.
I have tried to flash the module using a power supply in my rack and turn the power on and off 10-20 times after disconnecting the usb but still black display. I do not think this is due to power instability. I did the test using a 4ms raw power 25, only connected the O_c so it should have plenty of power.  eh2k If your module is diy. Which display do you use? It might be easier for me to by a new display, and play with 0.0l while I wait :)
Regards
Arnfris

eh2k, 2022-09-09:
Hello @Arnfris, @themoonscientistandco, @n0ahg 

I tested it with my DIY 7812/7912 power supply, the display always worked after various on/offs...

My guess is that it has something to do with the SPI interconnection, maybe the line is longer, resistor tolerances, I don't know.

I had a similar problem with my DYI Braids - an "original" BluePill somehow had a problem with the SPI, but a "clone" BluePill did not - with the same firmware.

@Arnfris - I bought my "1.3 inch SPI 128x64 SSH1106" from amazon 2 years ago - with white colour - I think it all comes from alliexcpress somehow.

I would like to try a few more changes though, I would appreciate your patience.

Trial no. 3:
   - https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/firmware_00M3.hex
     - reverted changes: 
        - Teensy platform 4.15.0 to 4.17.0 upgrade
        - Typo in the SPI Initialization - clock 4000000 instead of 400000.
        - -D_GLIBCXX_USE_C99 to -std=c++17
        - added back "yield()" calls after spi_transfer  (display driver)

If that doesn't work - I have another display lying around (smaller, without CS connector) - maybe I'll have to try to get it to work with that - maybe I'll get some insights.

Arnfris, 2022-09-09:
@eh2k @themoonscientistandco @n0ahg 
No problem at all. You are the one who is doing all the work. I will continue testing as long as it takes.

spiilmusic, 2022-09-09:
Hi. I updated all 3 of my s_c to the 0.0m firmware and have not found any problems with the display yet.

Arnfris, 2022-09-09:
@eh2k @n0ahg @themoonscientistandco @spiilmusic 
Hello.
Still black screen here.

themoonscientistandco, 2022-09-09:
Hallo , i flashed version 00M3 and the screen works perfect from the first time , I turn of the power and on again a few times and it seems my problem is solved ,00m3 is (for the screen problem) the best version for me
many thanks ,Grtz 
The Moonscientist

n0ahg, 2022-09-09:
Hi,

Still black screen here.  For reference my uOC is this one
https://pushermanproductions.com/product/full-kit-smd-pre-soldered-uoc-with-brushed-aluminium-panel-blue-screen/

Arnfris, 2022-09-09:
@eh2k @n0ahg 
Hello 
Mine is build on this pcb:
https://pushermanproductions.com/product/ornament-crime-pcb/
The screen , blue,pusherman.
I was wondering. Have you done the "blind test" n0agh? (Listen for sound while giving a trigger to a drum on black screen). I was thinking that if you can't hear sound maby we are dealing with two different ptoblems as themooschientist who heard sound has solved the problem with his module.

eh2k, 2022-09-09:
I'm glad the update is working as expected for some. 
Thanks to all for troubleshooting. 

The commonality I currently see at @Arnfris and @n0ahg is the blue pusherman display. 

Another idea I would have - connecting the display with an extension cable. If it stays black, I could keep trying around until it works.

FYI: Regarding pusherman - I know he sells OCs with Teensy4 and Squares&Circles. He has never contacted me - and I have no agreement with him.

Arnfris, 2022-09-09:
Hello. 
Yes, I think i will try to contact pusherman for a recomandation on screen. :)

Arnfris, 2022-09-09:
@eh2k 
Thank you wery much for all troubleshooting and the  work on the firmware :) :)
Refards
Arnfris

eh2k, 2022-09-10:
@Arnfris @n0ahg 

I have tested another display - a small ssd1306 (actually the same connections but without CS). I flashed the firmware 0.0m and the display is working (there is a expected 2pixel x-offset with the ssd1306).
[image]

I looked it up, this is the display i bought from amanzon 2 years ago (still available in germany): https://amzn.to/3Bvp0Z5

However, I am now unsure whether it is really only the display that is not working, or whether the Teensy hangs at start-up. If you can tell me clearly that the "blind/browser" test works, that would be good: [https://eh2k.github.io/□●/test/](https://eh2k.github.io/%E2%96%A1%E2%97%8F/test/)

n0ahg, 2022-09-10:
Hi,

I can't connect rack power and USB as I have not cut the trace on the
Teensy (I don't intend to use USB MIDI at this time and this makes flashing
outside the rack easier).  Is there a test I could perform without rack
power and USB power?

Thanks

eh2k, 2022-09-10:
@n0ahg for the test you only need the teensy connected via USB (powered by USB) - like in the video. If you have cut the USB power ist is also working if the teensy is placed in the OC (powered by the OC).

n0ahg, 2022-09-10:
Sorry, I was on my phone and misread the test as a rack sound test so
assumed I'd need rack power and usb connected :-/

I can confirm that when using the test page you provided I can connect to
the teensy and navigate around the menus.

Thank you for all the hard work

Arnfris, 2022-09-10:
@eh2k @n0ahg 
I now can confirm that i have sound on black screen on the 0.0m3 firmware. It most likely is the display then that is casing the trouble. (I must have done something wrong the first time i tested).

wtkdwc, 2022-09-10:
hi, just gave it a try too and with 0.0m3 black screen here too. 0.0l works fine. on the module i have this screen https://www.aliexpress.com/item/32818944937.html

eh2k, 2022-09-11:
Hello @Arnfris, @n0ahg, @wtkdwc 

A new day, a new attempt:

Trial no.4:
  -   https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/firmware_00M4.hex
      - reverted DigitalWriteFast to DigitalWrite display DC, CS

Have a nice Sunday

Arnfris, 2022-09-11:
Hello. 
it might loock like your har work has payed of. Mine is working :) Supercool :)

Arnfris, 2022-09-11:
@eh2k @n0ahg @spiilmusic @wtkdwc 
Once more( Forgot the tag)
Hello it loocks like your hard eork payed off.
Mine is working. If it is the same result for the others this is loocking real good. .supercoool :)

themoonscientistandco, 2022-09-11:
Good morning eh2K 
Even version 00M3 works fine for me, I flashed version 00M4 and it works super well
Again, thanks for your hard work
Respect
Grtz
The Moonscientist

wtkdwc, 2022-09-11:
@eh2k thanks, this version works for me too 🥇

MidiSlave, 2022-09-12:
had the same issues here but flashing 00M4 worked.

eh2k, 2022-09-13:
Hello @Arnfris @themoonscientistandco @n0ahg @wtkdwc @spiilmusic 

There is a new version: [image]

I hope this time without any after improvements, I will tag this as a release the next days.

Here is some background info:

One of the big challenges with this project was to get the display (20Hz) and the DAC (4x48Khz) to work the same SPI port. So far I have tried to avoid display updating, and was very sparing with "drawing", as with some engines there was audible interference. For example, most notably with the Rings resonator, you could hear clicks when you scrolled through the menu. So far this was the reason why I didn't implement any "waveform visualization" or visual moving sequencers etc...

All this is history with RC_0.0N (RC_0.0M). I have introduced a screensaver with the N version - actually as a demo of this optimization. You can activate it without waiting 5 minutes by pressing the "left button" for 2 seconds. 

Thanks again for troubleshooting.

Arnfris, 2022-09-13:
@eh2k @themoonscientistandco @n0ahg @spiilmusic @wtkdwc 
Hello.
Nice work :)
The webflasher and screensaver workes fine.
Regards
Arnfris

eh2k, 2022-09-22:
issue closed - fixed in https://github.com/eh2k/squares-and-circles/releases/tag/RC_0.0M4

closed 2022-09-22 by eh2k

n0ahg, 2022-10-11:
That's the one!  Great job... screen working here now using 00M4

Thank you