eh2k/squares-and-circles #19: Midi pitches no longer working

MidiSlave, 2022-09-12:
I'm on 004m now but found the same in 0.0l. When I play chromatically via usb midi the module is only recognising change of octaves. Triggering and CC's are working fine.

MidiSlave, 2022-09-12:
I guess while we're on the subject of midi. Program changes don't seem to work either. sending any PC to it changes the machine to GND

eh2k, 2022-09-12:
Hello @MidiSlave,

thanks for feedback.

I can confirm the same behaviour - ProgramChange ist not working through USB-Midi.

I'm using only the [usb_midi_read_message()](https://github.com/PaulStoffregen/cores/blob/master/teensy4/usb_midi.c#L303) function - it looks like you can't directly push the 4 bytes into the [MidiHandler](https://github.com/eh2k/squares-and-circles/blob/main/lib/stmlib/midi/midi.h#L64) in some cases. 

usb_midi_read_message() returns `0c c0 02 00` that results in
```
ProgramChange 0 12
ProgramChange 0 2
ProgramChange 0 0
```

eh2k, 2022-09-13:
@MidiSlave,

The issue should be fixed in the latest version: [image]

MidiSlave, 2022-09-13:
The web flasher doesn't work for me. Using chrome on a macbook. "NotAllowedError: Failed to write the report.-" But after loading 0.0N, Program changes **do work** now! however midi pitches still **do not**. Chromatic pitches played via USB keyboard only plays in octave intervals.

eh2k, 2022-09-15:
Hello @MidiSlave,

thanks for the hint.

This is a bit complicated - at the moment there is different behaviour depending on the engine. For example, the "peaks" engines, like "FM drum" engines, generally cannot pitch. Engines based on "braids" cannot gate, i.e. trigger at NoteOn, NoteOff doesn't matter. There is currently a bug that only triggers the notes octave by octave with mono engines (I will fix this). With the "midi/VAx6" the semitone notes should work, but the pitchbend is not yet implemented (https://github.com/eh2k/squares-and-circles/blob/main/src/polyVA.cxx#L117).

Regarding pitch-bend - the range is not standardised (and would actually have to be configured). I would set this to fixed +- 1 octave in the next release (currently +- 1/4 octave).

Somehow the pitch-bend does not seem to be transferred correctly via usbMidi. I don't know exactly where the problem is - maybe a bug in the teensy core lib.

I have built a test frontend [here](https://eh2k.github.io/□●/test/?midi) (I don't have a usb midi keyboard so far), if you like, you can test/verify it with the "midi/monitor". It would be helpful to know, if the behaviour specifically the pitch-bend doesn't work at you either or is different compared with your keyboard.

Regarding the problem with the "web flasher", could you please make sure that the "TeensyLoader" is not running in the background.

MidiSlave, 2022-09-17:
I've since reverted to 0.0k where midi is working fine and no freezing.
Had a gig the other night. Are you asking load 0.0N on it and test with
midi/monitor?

eh2k, 2022-09-17:
Hello @MidiSlave 

I have fixed/implemented these in the 0.0N2 Version [image]
  * USB keyboard only plays in octave intervals on `M-OSC` Engines
  * `Drum/FM-Drum` pitch controllable per CV/Midi
  * `Midi/Monitor` executes midi-reset in constructor (midi panic)
  * Midi Pitch-Bend range = +- 1 Octave
  
I am not aware of any other midi issues with 0.0N2 now, except that some engines (808-...) do not support midi pitch.

If there are still problems, it would be good to reproduce them on the [Web-Midi-Test-Page](https://eh2k.github.io/%E2%96%A1%E2%97%8F/test/?midi), so that I can rule out whether it is due to the keyboard or the control hardware.

PS: If you have any videos or recordings from your gig, it would be great to see/hear them.

MidiSlave, 2022-09-17:
awesome. Chromatic pitches are working again but the machines are quite out
of tune. eg. I've transposed the first machine -17, 2nd  -19 and 3rd -24 to
turn on semitone quantization on to get them in tune.

eh2k, 2022-09-17:
@MidiSlave 
 
hmm - I think we're kind of talking past each other.

I suspect, however, that perhaps the ADCs are not calibrated correctly.

The whole thing is a little complicated, but perhaps the description of how the pitch is calculated will help:

The pitch is made up of the following variables:
  * CV Input: 1V / Octave (quantized)
  * optional modulation (quantized)
  * Transpose offset: -24 = - 2V  = -2 octaves (default -24 is because better for baselines)
  * Midi pitch (+- 1 octave)
  * MidiNote C4 aka Note 60 = 0V

Please deactivate CV input by setting CV to "-" in the IO-Config. Consequently, the pitch is only controlled by Midi - everything should be in tune. C4 (Note 60) is equal to a trigger with 0V at the CV (CV not connected). 

If something is not as expected, please name the engine you are using  (e.g. M-OSC/Waveforms), so that I can maybe recreate it.

MidiSlave, 2022-09-18:
> Ah yeh deactivating the cv inputs!
>
> We’ll it’s working pretty solid so far. See video. I’ve sent an lfo to the
> decay cc to keep updating with a low value for the drums.
>
> Such an incredible bit of gear. Hats off

https://drive.google.com/file/d/1j8NMA-DjWbfRZA5mozB6AO0LZg1zyyP1/view?usp=drivesdk

eh2k, 2022-09-22:
@MidiSlave, thank you for the video.
It's really crazy what you're doing - are you actually using the Nerdseq to control it? - actually it's nothing different than the paramter locks on the electron boxes :-)

Do I get your permission, I would like to directly embed the video in the 0.0N4 release notes ?

I'm closing this issue, if there's anything else that stands out with the 0.0N4, please just create a new issue.

MidiSlave, 2022-09-22:
Yeh, go for it.
But no, not in this instance. T4 is being sequenced from Digitakt which
allows PC message locks per step. Octatrack is sequencing T1-T3 (on same
midi channel) for the poly-arp.