eh2k/squares-and-circles #20: Screenatifacts when returning from screensaver RC_0.0N.

eh2k, 2022-09-14:
Have a little bit problems when returning from screensaver.
[image]

_Originally posted by @tsanderakis in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1247062816_

eh2k, 2022-09-14:
Hello @tsanderakis, thanks for reporting the issue.

The screensaver can be activated by pressing the "left button" for 2 seconds, without waiting 5 minutes.

Does this happen every time, or only once by chance? Could you please make a short test series and tell me how often it happens. For example, in 5 attempts on your device?

tsanderakis, 2022-09-14:
Don`t know which Firmware I have, but for me it is not possible to switch
on the screensaver by left switch and also tried to press the encoder
switches. In the moment I can`t find a system behind. 10 minutes ago the
komplete sound was off. After starting the rack new everything was fine
again. We`ll keep in touch and I`ll let you know if I can find something.
My PSU should be strong enough.

eh2k, 2022-09-14:
@tsanderakis, 

You can see the version number in the startup screen.
If you do not have the current version, please upgrade to the latest [image] version.

eh2k, 2022-09-16:
> [image]
Wednesday 3 hours and yesterday 10 hours torturing the firmware 0.0N with Mutable Grids. On wednesday two times I had hanging pixels in same place. Screen refresh? Yesterday in these 10 hours everything was perfect. No frozen engine, no pixels und for now it`s looking good. Have only tested the drum department.

> _Originally posted by @tsanderakis in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1249040519_

Many thanks for the report:

10 hours without failure is not bad. However, the fact that it occurs despite this is not good. I would perhaps observe this for a while - maybe you can somehow simply reproduce it.

tsanderakis, 2022-09-16:
> (comment)

Ok, will do another testrun. So what was confusing is 2x in 3 hours and
next day working perfect 10 hours without any problem??

tsanderakis, 2022-09-16:
What I can also do, but maybe in a few days, is to change the module. I`m
in this lucky situation that I have 3 of these O&Cs.

eh2k, 2022-09-17:
> 9 hours test with 60 bpm. Running perfect without any problems. I will give it a second try with 200 bpm this afternoon.

_Originally posted by @tsanderakis in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1250020708_

Hello @tsanderakis,

thanks for testing.

I don't think it matters how many bpm it runs at.
However, it would certainly be interesting to know if the problem only occurs on a specific OC or if you also experience it on several devices.

grtz
eh2k

eh2k, 2022-09-17:
>[image]
New O&C Hardware. After a little bit more then 1 hour and 170 bpm same problem.

>_Originally posted by @tsanderakis in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1250068074_

Hello @tsanderakis,

thanks for reporting. I read that the problem occurs on two different 14HP O_Cs.
I have made a few adjustments in the 0.0N3 Firmaware [image]. Could you please flash it and let me know if it you have still any issues.
Thanks in advance

eh2k, 2022-09-17:
>With the new version my Teensy stays dark. No reaction. Panic. First I was thinking I killed it, but when going to the older firmware everything was working. Giving a second try but looks the same like before. Screen stays black and no reaction. Do I have to do something before flashing with these 3 pushbuttons at the bottom of the flashing page?
>_Originally posted by @tsanderakis in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1250103583_

Hello @tsanderakis,

hmm, thanks - give me another try:

I have reduced the SPI frequency in the RC_0.0N3, somehow there seem to be problems with the initialization. Could you please test this version again [image]. 

To avoid misunderstandings, could you please name the exact firmware version you are using? I am otherwise unsure of the exact version is used.

Thanks again

eh2k, 2022-09-18:
Hello @tsanderakis,

a pity it still occurs. Could you please test it with [image]. I hope that it will work without issues now.
Can you tell me if the display works well when you scroll through the menus etc.?

Many thanks for troubleshooting

>[image]
0.0N4 with 170 bpm
After pressing left encoder it is looking like this

>[image]

>_Originally posted by @tsanderakis in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1250128729_

eh2k, 2022-09-18:
Hello @tsanderakis 

you wrote that you have several O_Cs. Did you use the same Teensy Board? What actually happens after the display has this artifacts, and you open the machine menu, is everything then good after that?

Please try to reply in this issue, somehow you always reply in the general feedback thread. Thank you.

> 0.0N5 3 hours with 170 bpm
[image]
[image]

_Originally posted by @tsanderakis in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1250225643_

tsanderakis, 2022-09-18:
The teensy board 4.0 is the same. I changed the O&C module. No. 1 and No 2
testing with ~same result. I bought 2 teensys 4.0 a few weeks ago and if
you give me a little bit of time I try to find the second one. I put it
away in a save place but can`t remember where and in which box. Hardware
problem with teensy?

eh2k, 2022-09-18:
@tsanderakis 

I'm not sure at the moment whether it's just the display that's having a problem or whether the module is generally crashing or hanging. Can you tell me whether sound comes up or whether you can navigate in the menu afterwards and whether the display corrects itself?

if the module hangs, can you tell me the engines you are using?

tsanderakis, 2022-09-19:
Fresh Teensy 4.0 out of the bag  Software 0.0N 200 bpm. After 1/2 hour
completely frozen and no chance to wake it up by pushing encoder.
[image: DSC08718.JPG]
Changed the module and after 10 min same problem. Audio out frozen but it
was possible to go back to the menue, but does not show up correctly.
This time I also used the right encoder and now I can see a "moving menue"
from left to right. Looks like the menue is not scrolling smoothly but a
little bit more jumping. And I can start and stop the scrolling by pressing
right encoder.
What do you mean with engines?

Later this afternoon I will take a different PSU and do another test.

eh2k, 2022-09-19:
@tsanderakis,

> Fresh Teensy 4.0 out of the bag  Software 0.0N 200 bpm. After 1/2 hour
completely frozen and no chance to wake it up by pushing encoder.

ok, then we have a completely different problem here, when it freezes and nothing else works (no menu, no sound).

When this happens again, could you please try pressing the [right encoder] & [right button] at the same time - and tell me if the reset works.

If the reset doesn't work, then the unit is really hanging. It is best to use firmware [image].

With the engines I mean e.g. "Drums/FM-Drum". Unfortunately, I don't see which other engines are loaded.

Thank you.

closed 2024-01-16 by eh2k