eh2k/squares-and-circles #21: Channel one gate input not working (hardware works with standard OC firmware)

BAREBONESLDN, 2022-09-16:
I've tried several teensy 4's and reflashed the firmware several times (with the two latest versions) but for some reason channel one is not responding to the gate signal. 
I can get audio out of channel one by using the speech synthesis mode.
The gate input tests fine in the testing menu and works correctly with a teensy 3.2 and standard OC firmware.
Any ideas what could be going wrong?

Thanks Phil

themoonscientistandco, 2022-09-16:
hallo , barebonesldn ,your not alone ,I have the same problem , first i thought it was my fault (clumsy smd soldering)
but on the test page everything seems ok
now that you have the exact same problem on trig input 1 i'm starting to think it's not a hardware problem
grtz
the moonscientist

eh2k, 2022-09-16:
@BAREBONESLDN @themoonscientistandco 

Please check the Midi-Settings by pressing both (L+R) Buttons.
[[image]]

Propable the Midi-Input is "T1" and the port is initalized as "Serial Input".
If it is "USB" so try to change it to "T1" and back to "USB".
After that it should work. Save the Preset with both (L+R) encoder pressed.

It has happened to me before, but I haven't figured out how to reproduce the condition.

themoonscientistandco, 2022-09-16:
Hallo eh2k . indeed, that was the problem
thx
grtz
the moonscientist

BAREBONESLDN, 2022-09-19:
Unfortunately this fix did not work for me.
I also tried installing the hew firmware update but no difference.

eh2k, 2022-09-19:
@BAREBONESLDN,

Please press [left button] while startup, and goto the IO-Test page - there you can test whether the input generally works.
[image]

BAREBONESLDN, 2022-09-19:
I figured it out it was the trigger input setting on the channel, it was set to -
Hold down right button to enter the sub menu for inputs and channel gain.

closed 2024-01-13 by eh2k