eh2k/squares-and-circles #23: Freezing when using USB MIDI

303sec, 2022-10-20:
I’m connecting a DIY data-only USB cable from my Deluge to a uO_C on S&C (0.0N4) but after a random period of being used it will freeze and need a reboot. It doesn’t seem related to amount of processing being used. Are there any crash logs? Or any way for me to try and debug what’s going on?

eh2k, 2022-10-21:
Hi @303sec, thanks for reporting.

I am sorry to confess that I don't really use USB midi and therefore haven't tested it well. With midi via TR1 this has never happened to me. It could well be that it is generally due to the USB midi. It also seems very strange to me that the teensy freezes - usually when it crashes it reboots. It could be that many program (engine) changes cause memory fragmentation. 

It would help if you could reproduce the problem somehow. Maybe by running it through a PC and recording/logging the midi stream.

303sec, 2022-10-22:
Yeah I’m using a standalone controller too (Synthstrom Deluge) which has been a bit weird with midi in the past… I’ll do some experimenting and see if I can replicate the freeze with a PC and then log the midi stream.

Any suggestions for recording midi streams on Linux?

eh2k, 2022-10-22:
@303sec,

have never really done it - except "--dump", but i think "send" "receive" could be helpful for this: https://linux.die.net/man/1/amidi

eh2k, 2023-03-19:
@303sec,

I have a new version here [https://eh2k.github.io/□●/flash/?ver=@28_0.79](https://eh2k.github.io/%E2%96%A1%E2%97%8F/flash/?ver=@28_0.79) - there I have optimized some things regarding midi - can you please give me some feedback if you can reproduce the problem in this version.

eh2k, 2023-03-26:
Issue goes to close - same issue as https://github.com/eh2k/squares-and-circles/issues/28

closed 2023-03-26 by eh2k