eh2k/squares-and-circles #25: OLED 1 Pixel offset...

eh2k, 2022-11-18:
How do I adjust for the offset of the OLED?
If this is not a feature, it would really help.
My screen is like 1 pixel off and looks bad.
Could do this on other firmwares, so hope to see it added or explained to me how to adjust.

Thanks!

_Originally posted by @ehafh in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1319329208_

eh2k, 2022-11-18:
@ehafh,

could you please give me more information about your display. A picture or a link of what exactly is installed in your device ( uO_C? ) 

Sounds like an SSD1309 display, instead of SH1106.

ehafh, 2022-11-18:
Hello,
I am positive it's an SH1106.
I put a Teensy 3.2 on the module, and same issue with being offset. 
But the stock firmware lets you shift the graphic by a couple pixels.
This only happens with this OLED from some providers, I have built dozens of the module.
Basically, some of these screens are different.  If you look at notes about the O&C from the uOC page, it mentions having wider holes to accommodate for these other variants. Sometimes from Amazon you get this type, sometimes you get the other. Or from eBay or Aliexpress etc same thing in my experience.

https://github.com/jakplugg/uO_c
"Note about the panel: there are 2 different hole spacing for the displays, Adafruit or ones made from the adafruit gerbers will fit the FPD panel in the panel folder. chinese ones from ebay, amazon, alliexpress, etc have a different hole spacing (see the pdf for oval holes used on PCB-panel) the PCB also has oval holes to support both displays."

Here are some example photos. I don't have the other variant of the OLED on hand at the moment, just this one.

https://i.postimg.cc/jwm6fmQq/20221118-111304.jpg

https://i.postimg.cc/tn63Nk23/20221118-111354.jpg

https://i.postimg.cc/vDxty8b4/20221118-111408.jpg

https://i.postimg.cc/sBn4jvMh/20221118-111415.jpg

eh2k, 2022-11-19:
Hi @ehafh,

Thanks for your pictures: For me it looks exactly the same when I connect the small 0.96" SSD1603 display.
Here is a good background info: https://forum.arduino.cc/t/whats-the-difference-between-adafruit-ssd1306-and-sh1106/247368/5

could you please install this firmware (https://eh2k.github.io/□●/flash/?ver=@25), and then use the "setup command" to switch between SH1106 and SSD1306 and report back to me if the "SSD1306 mode" is fixing your issue.

[image]

Thanks in advance

ehafh, 2022-11-19:
Wow thanks for such a quick update. That fixed the offset for me!

I will eat my words about being positive it's an SH1106 controller haha.
Just looked up the listing I purchased it from and it doesn't state which one it is.
So you solved the problem. Thanks again!

If it helps you out I can mail you one of these screens, just let me know :)

eh2k, 2022-11-19:
thanks for your response - that's great - I will close this issue when this is included in the next release.

closed 2024-01-13 by eh2k