eh2k/squares-and-circles #28: Drum Engines causing freeze

MidiSlave, 2023-01-23:
I've had a couple gigs recently and have had to abort my plans to use Squares and Circles for drum voices because it freezes. I can have it on M-OSC for hours with out it freezing but almost with out fail will freeze if using the drum voices and perhaps especially the vintage hats/rom machines. I've been sequencing/controlling it via midi. Still love the hell out of it using as polyphonic M-OSC!

eh2k, 2023-01-23:
@MidiSlave 

Thanks for reporting.

I assume you are using UsbMidi?
If possible, can you provide me with a midi file so I can try to reproduce the freeze?

I suspect this has something to do with memory fragmentation. As I saw in your video, there is extreme fast engine changes. With an unfortunate sequence, the memory could theoretically run out.

I would have to think through the memory allocation, but I don't know if it's a different Issue.

See also https://github.com/eh2k/squares-and-circles/issues/23

MidiSlave, 2023-01-23:
Yes using USB midi. I’ll try via my computer/midi and send you a midi file,
however the freezing seems to occur when using drum engines regardless of
fast program changes being sent to it or no PC messages at all.

eh2k, 2023-03-19:
@MidiSlave 

I have a new version here https://eh2k.github.io/□●/flash/?ver=@28_0.79 - there I have optimized some things regarding midi - can you maybe give me feedback if you can still reproduce the problem in this version.

MidiSlave, 2023-03-26:
> I have a new version here ***@***.***_0.79 -

awesome, i'll give it a go. do you think the new plaits firmware voices
will make it onto this?

MidiSlave, 2023-03-26:
No problems so far with the freezing :) but the programs numbers seem out
of order compared to the machines menu.

MidiSlave, 2023-03-26:
Actually....several freezes and now the screens off when i reboot.

eh2k, 2023-03-26:
HI @MidiSlave,

thanks for the response - I have opened a new issue regarding the program numbers.

I have now tested this a bit more intensively. If I simply set random program numbers (0-44) in a loop, I can reproduce the freeze now.

However, if I avoid the program numbers 33,39,40,41 ( FX engines that allocate a lot of memory ) - it actually works without freezing. For completeness - here is my test-script: https://github.com/eh2k/squares-and-circles/blob/main/test/test_midi.sh

I'll have to take a look, maybe I'll come up with something else.

Can you please give me more info about what exactly you do - especially which parameters and which engines you use.

PS: As far as I know, the recently published plaits update is not open source - but there are some old engines like Swarm, Noise, Particle that I could add at some point.

MidiSlave, 2023-03-27:
I have a midi track on digitakt sequencing the first channel of
squares-and-circles  - Alternating between (p-locked) 808ish-BD (PC#7) and Clap engine (PC#32) on midi channel 9. on the next midi track hihat (PC#32) on midi channel 10 and another on channel 12 for VA6 engine. The only other parameter other than the PC messages being sent is CC7 for volume. I'll try removing the volume automation and see if that helps.

closed 2024-01-14 by eh2k