eh2k/squares-and-circles #29: GM Standard Drum Map

eh2k, 2023-01-28:
>... An idea is to expand the ROM drum sounds if there is flash left. The individual sounds in that engine could be triggered via MIDI notes general MIDI style. ...

_Originally posted by @rheslip in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1399544859_
            

@rheslip

thanks for your feedback and suggestions.

I have already thought about the implementation. 

One possibility would be to simply introduce a new engine e.g. based on the TR-707 similar to Midi/VAx6 in the Midi category. 

Another possibility would be, in the Midi settings somehow handle the Channel-10 separately, and there assign individual keys to different Engins.

https://de.wikipedia.org/wiki/General_MIDI#/media/Datei:GM_Standard_Drum_Map_on_the_keyboard.svg

eh2k, 2024-10-29:
@rootapprox 

That was the first thing that came to my mind after I had finished the multi-trigger. 
It's next on my todo list. I will add the parameters in the IO-page under the pan and level parameter, just like with the trigger.

rootapprox, 2024-10-29:
that would be crazy dope!
i also would love to see some kind of modulation for the PAN in normal channels , like you can add an LFO for the pan maybe.

Very Great work! 
also if there is a way to be notified each time there is a Update ! 
cheers

eh2k, 2024-10-29:
@rootapprox - check out the latest version - you can set level/pan for sound, where trigger is assigned.

yes, modulations would probably be the next thing missing, unfortunately i have a few limitations that i can't easily break.

I'm currently in a test/fixing phase where I'm getting some feedback from you and others. I try to push fixes as soon as possible. In a few days I will tag the latest commit as a release version and will close all fixed issues. If you starred/subscribed the repo or issue, you should get a notification...

closed 2024-12-01 by eh2k