eh2k/squares-and-circles #31: MIDI via T1

eh2k, 2023-01-28:
> I've tried MIDI input on T1 and so far its not working for me. ...

_Originally posted by @rheslip in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1399544859_
            
@rheslip

thanks for your feedback.

Do you have more info on how you connected it exactly?

Personally, I actually use midi only via the TR1 jack (midi via usb I have used something via the PC, but have no long-term experience).

For this I used the circuit from the readme. But I think it would be easier to implement, as long as the transister Q1 switches.

[image]
-

rheslip, 2023-01-28:
Thanks for the reply.

I'm using an Arturia TS to MIDI adapter which has the + side of MIDI on the tip. I just checked it on the bench and its able to turn on the transistor in the O-C circuit below. The T1 input on O-C is definitely working. So I assume my problem is the software setup.

On the MIDI setup page I have T1 as MIDI in and I've set each of the MIDI-CH#s to 1. I have verified that MIDI is on channel 1. I've tried various settings for trigger in using the resonator engine and I get nothing. If I put an signal on T2 LFO and assign trig to T2 it works. I see nothing in the MIDI monitor engine either.

Am I missing something in the setup?

Thanks, Rich

Originally posted by @rheslip<https://github.com/rheslip> in #1 (comment)<https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1399544859>

[https://github.com/eh2k/squares-and-circles/raw/main/doc/midi2ts.png]<https://github.com/eh2k/squares-and-circles/raw/main/doc/midi2ts.png>

-

eh2k, 2023-01-29:
Sounds like the same issue #31 .

I just tried this on my device - and it works. You have actually done everything right so far:
 * `Engine #1`is assigned to midi channel 1.
 * The midi monitor is running on `#1`.

The only thing you could try is to change the midi input from T1 to USB and back again (there were reports that it somehow doesn't always re-initialize after rebooting).

Otherwise, I suspect that somehow the connection does not fit electrically. Can you tell me exactly which adapter you are using.

eh2k, 2023-01-29:
I think that a type-c TS adapter unfortunately cannot work. See also discussion https://github.com/eh2k/squares-and-circles/discussions/33

closed 2023-01-29 by eh2k

rheslip, 2023-01-30:
The Arturia adapter (from my Beatstep) is actually a TR. I set the whole thing up on my bench and as you can see there are negative going pulses on the output of Q1 (TR1) on the O-C mini. Tried switching back and forth from T1 to USB input, no joy. USB MIDI does work. If this works for you it should be working on my setup but it doesn't.

[image]
[image]
[image]

eh2k, 2023-01-30:
I looked at the connections of the TR adapter yesterday, and as I understand it, it can't work.

https://minimidi.world/?fbclid=IwAR31TqOyRkvdwaLYCxoU2a89hcy2PF3hltCtRKD7IzD5HbZqzn3m9NmiZzc#types

Please correct me if I am wrong:

[image]

The TIP is connected with PIN 4 - and is permanently 5V.
The SLEVE is connected to PIN 5 - that is basically a pull-down, that is connected to the TX GPIO (e.g. https://github.com/eh2k/mutable-instruments-diy-archive/blob/main/docs/static/schematics/Shruthi-Digital-v08.pdf) 

Since the transistor Q1, when it switches, pulls the Teensy pin 0 (TR1) to ground - the serial signal must be read inverted. Could you test if the level at the collector or Teensy PIN 0 changes also between 0 and 3.3V - as I see you are measuring the base of the transistor?

rheslip, 2023-01-30:
You are correct - a TR adapter will not work regardless of how its wired up because the O-C input inverts the MIDI signal. 

Thanks!