eh2k/squares-and-circles #35: Voltage mod means I cannot calibrate properly

BAREBONESLDN, 2023-03-09:
Hi,

I did the voltage mod which works fine but means I cannot complete the final stage of the calibration as it's now out of range.
Should I calibrate first and  then add the extra resistor for the voltage mod?

Thanks!

eh2k, 2023-03-09:
Hi, you should definitely calibrate after the mod. If you turn the encoder while holding it down, you can first roughly adjust the offset. For fine adjustment, turn the encoder without pressing it.

jpnielsen, 2023-04-03:
[image]

But how can the calibration routine know if we changed the resistor or not ? 
I will try to think and experiment more with this - I still haven't done the resistor mod.

Could we have two different calibrations - resistor mod on and off ? In case we want to put a switch on the resistor mod.