eh2k/squares-and-circles #38: T&H Modulation - wrong Trigger

eh2k, 2023-04-10:
Hello eh2k,

uploaded yesterday the latest 0.0N4 firmware and I'm using it on a hardware that was hosting Hemisphere and Benisphere without problems.
I'm still figuring my way around it a bit, but I seem to have come across a problem with the T&H CV modulation. Basically what happens is that the triggers are shifted by one value, so:
- Op: T&H-T1 is triggered by T2
- Op: T&H-T2 is triggered by T3
- Op: T&H-T3 is triggered by T4
- Op: T&H-T4 is triggered by nothing

Am I missing something or is there an actual problem?
Thanks, enjoying it very much so far!!
N.

_Originally posted by @lancerotti in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1499180396_

lancerotti, 2023-04-10:
Very nice!

Another doubt I have is concerning  the "color" knob in the "Plaits" section. It doesn't seem to be doing much, at least to my ears.

Thank you for all the great work
N.

eh2k, 2023-04-10:
@lancerotti

Color is the internal `lpg_colour` paramter and changes the sound in the higher frequency range - if you have something low-frequency, this is not so well audible. You can hear it best with higher notes and with the chord enigne.

lancerotti, 2023-04-10:
Thanks again,

I'll give it a better listen as soon as I can.

N.

closed 2024-01-14 by eh2k