eh2k/squares-and-circles #42: Other drum machine roms

waldnzwrld, 2023-05-21:
Out of curiosity  would it be possible to say grab an hr16 rom dump and use it 

Or maybe a linndrum or ensoniq sq80
?

waldnzwrld, 2023-05-21:
Just asking since theres a fairly large archive of vintage synth roms

eh2k, 2023-05-21:
Hi @waldnzwrld,

I watched a HR16 demo on youtube - I didn't know the drum machine before. 
Shouldn't be a big deal - since it's just 8-bit samples, similar to the Linndrum and DMX. 
The original ROMS can also be found at https://dbwbp.com/index.php/9-misc/37-synth-eprom-dumps.

For the SQ80, I see it's just patches (similar to DX7) - i.e. these can't be used so directly, you'd need an emulator for the synth.

I recently reworked the sources and removed the TR707 or TR909 samples - I'm working on a way for the user to then flash these ROMs independently of the actual firmware.

If you have any tips for interesting old hardware (like HR16), just post them here.

waldnzwrld, 2023-05-21:
Well here is an archive kept alive mostly for old instrument repair

https://dbwbp.com/index.php/9-misc/37-synth-eprom-dumps

The hr16 was one of the great dm's of the early 90's dance sound

eh2k, 2023-05-29:
After some research, there are two sound eeproms for the HR16B which contain 49 samples.
These are raw 8bit, but can be decoded to 16bit.

With 2 x 512KB = 1MB sample data  already quite a bit for the Teensy 4.0 (2MB Flash)...

-------
>FROM: https://www.nekosynth.co.uk/wiki/AlesisSamples/ (https://www.burnkit2600.com/diy-sound-roms/)

#### Method of Decoding
Decoding the sample should be fairly simple. Let's assume we have a couple of helper functions that load the raw ROM image into an array, and write an array of floats out as a standard sound file (for instance, .WAV).

We need to read through the sample, and divide the output by two every time we hit a zero (and skip the zero so it doesn't get output). To do this, we might use something like this:

```c

// output multiplier, start off at unity gain
mult = 1;
// output pointer, lets us skip zeroes
outptr = 0;
// loop through each byte
for (i = 0; i < SAMPLE_SIZE; i++) {
  // get the sample
  s = input[i];
  if (s == 0) {
    mult = mult / 2;  // drop by 6dB
  } else {
    // put it in the output array, and bump the pointer
    output[outptr] = (1-(s/128)) * mult;
    outptr++;
  }
}
```
Note that this does not attempt to find the start and end of samples, or deal with any "odd bytes" that some samples seem to have. If this decoder produces correct results for the "factory" Alesis samples, it may be used to confirm that the encoder software is doing the right thing.

jpnielsen, 2023-07-22:
https://reverb.com/software/samples-and-loops/reverb/3514-reverb-drum-machines-the-complete-collection

these are sampled old machines, not ROMs