eh2k/squares-and-circles #43: .hex firmware

pppier, 2023-06-02:
hello, tnx for your work, giving new life to uoc hardware with teensy4 is a great thing
and i love teensy as it is so easy to flash for everyone, using .hex files, 
unlikely online downloader does not work for me (I tried with 2 different computers...now i'm using the last .hex file i downloaded) 
So I guess if there is a specific reason not to give the .hex file, as I think it would be a way to make things easier
tnx
p

eh2k, 2023-06-02:
hi @pppier,

thank you for reporting the issue.
It would be very helpful if you could provide more information. Screenshots or any error messages.

Actually, flashing via Chrome on Linux/Windwos/MacOS should work. I have tested it on various hardware and operating systems. However, my experience is that the TeensyLoader, when running in the background, interferes the flashing via browser.

Why I don't provide the HEX directly here:
 
In the past, I noticed that the firmware hex was shared on reddit (The posts show up pretty high in the search engines). For my part, it's a pity when some users don't even come to my github page and I can not prevent them of using old firmware versions with bugs that was long been fixed (in my spare time). 

I am currently in the process of moving some things out of the firmware. Specifically sample based engines need to be "installed" in a second step independently of the firmware update. I'm also leaning towards implementing more browser-based configuration options in the future. So I want to focus only on the browser way - and make it as stable and well working as possible.

Since you can also build the firmware yourself, it is actually no technical hurdle to get the hex file. 
Actually you can take the HEX file from the CI job (see Actions tab - Artifacts of the last build).
Please note that this is for personal use only.

Since we are on the subject, what do you think about flashing via audio (basically like all MI modules) ?
It would take longer, but would already be more practical, because most people cannot easily reach the USB port and have to unscrew the module to update.

pppier, 2023-06-03:
tnx for your fast reply
sorry, my message was too short: I love eurorack since 2000, i spent last years building for friends, as hobbist, hand soldering, (aka RTFM on modulargrid for example). I successfully built and calibrated some uoc that i find great to build and use, easy to flash and inspiring. 
the teensy3.2 issue is a pity in my opinion so when you released a firmware for teensy4 i was really happy, as i think i can make again this awesome module. (just two words on temps utile too, i think a teensy 4 version, somehow, would be a great thing too, like the midi signal on trigger input and so on ...)

I don't feel comfortable in flashing stm for example, and I find a low experience user can happily menage teensy and eventual upgrades, as we used to do in mxmxmx standard, as they released .hex file and all worked fine.
i personally am not interested in copying, stealing work or similar, I simply look for beautiful modules to build, following the scene. I like to give the designer what he asks for example (I never "produced" myself pcbs or panel even uoc was open source). i know arduino sketch environment but I have not the target to customize, or modify the firmware, as builder i just look for a safe and easy way for final users.

about firmware online flasher, my aim is to be able to menage the flashing easily, considering people, me on top, have not always the last browser upgrade. personally, I use a 2009 macbookpro running 10.11.6 (el capitan - until i can use paypal I don't plan to upgrade the system as it is alwasy a pita). before writing you i tested flashing a new uoc using the suggested method on:
- firefox (78.15), page is visualized correctly, but the module is not recognized so i have nothing in the menu. even "flash" and "reset" do not work. no link/button on "setup commands" area
- safari (11.1.2) page does not visualize the module image, it looks not to see the module when connected, "reset" and "flash" do not work. no link/button on "setup commands" area
 - google chrome (103.0.5060.134 - it tells me to get x10.13 to upgrade. i can successfully use the 16n online setup), no image, no "reset" or "flash" job, nothing on menu. "setpu commands" area shows button but i have no use of them
(if i try teensyloader, that was not in background, module is recognized and i can flash as usual) 

then I went to a friend's windows based pc, upgraded compare to my pc, it recognizes the module, i see 2 options on the menu but i cannot "flash" or "reset" ot use the "setup commands" buttons successfully. 

so i decided writing you asking for a step backward, or anything you think can be useful to make the module available again to be built and used easily
tnx again
[image]
[image]

daltronbeats, 2023-06-08:
@pppier - try flashing a blink sketch to the teensy first using the standard Arduino IDE. I just received 2 x teensy 4.0 and neither worked with the web flasher until there was at least some working sketch on the brand new teensy

eh2k, 2023-06-08:
@daltronbeats - thanks, that is a very good hint that would explain the issue. I can remember that the GUI TeensyLoader also updates the bootloader in the background. The web-flasher is basically a javascript port of the flashing process from the CLI version (https://www.pjrc.com/teensy/loader_cli.html). I still have a fresh Teensy here, I will try to reproduce the issue.

eh2k, 2023-07-22:
Here is a workaround/solution, If you are not able to flash the firmware on "bare" teensy:

Flash this HEX file with the TeensyLoader: 
- https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/squares-and-circles-teensy4XX-loader.hex.

This is a version of the firmware that has no engines, but enables you flashing the firmware by midi via browser.

After this firmware is flashed, please use the [image] to update to the full firmware with optional engines etc.

I have tested it with several browsers and also smartphones.
It seems that the midi interface is widely supported on many browsers and devices, rather than the serial interface used by the teensy bootloader.

closed 2023-07-22 by eh2k

pppier, 2023-08-04:
thank you for your time
yes, chrome. v 103.0.5060.134
(the teensy4 was flashed by the old firmware .hex so my issue was not related to "first flashing")

i flashed the base .hex, with teensyloader: all ok
but about webflasher, still no change 

how i can use the webflasher via midi ? what do you mean?
(i'm sorry for dumb questions, but i think final users can face this scenario too)

eh2k, 2023-08-04:
@pppier 

Thanks for your response - I hope the following information helps you, for me it really looks like an isolated case.
Unfortunately I can't do anything with the specified Chrome version, and would generally recommend using the latest version.

However:

- Please make sure that the Teensy Loader is not running in the background when you are using the WebFlasher.
- If you are flashing a bare teensy, the WebFlasher needs to communicate with the teensy bootloader via serial interface. 
   - The serial interface is not well supported by browsers due to security restrictions (only successfully tested with newer Chrome versions on Windows/Linux).
- The "updating process" and the flashing/installing of some engines is not handled by the "teensy bootloader". Here the communication between the squeres-and-circles firmware is handled directly via the well supported web-midi interface.
  - You can see the progress also on the o_c display (tested with Chrome, Firefox, Brave, also with my smartphone)
- In both cases, you need to connect the Teensy to your computer or smartphone with a USB cable.
  - The WebFlasher detects the installed squares-and-circles version, and is handling the further interface selection automatically.

Hope this makes the process easier to understand. Please keep in mind that the firmware is "modular" and there is not really one HEX file that contains everything.