eh2k/squares-and-circles #44: DxFM Data

daltronbeats, 2023-06-06:
Great work on the firmware! any hints on the DxFM implementation? what is the data its expecting? sysex over MIDI? an audio playback / tape thing?

or have i missed something and I need to add an SD card?
Thanks again!

eh2k, 2023-06-08:
@daltronbeats, 

It is actually "work in progress". Parameter control, polyphony, and some documentation is still todo...

Patch uploading is also working with the [Plaits-Editor](https://pichenettes.github.io/plaits-editor/patch_bank.html).

Feedback is welcome.

daltronbeats, 2023-06-09:
thanks! how long should the 'loading patch.' screen take? I've tried with a single patch and a full bank with varying gain but it seems to always hang on this screen. if i turn an encoder the module reboots and no patches are loaded.
Thanks again for your great work on this

eh2k, 2023-06-09:
Hi @daltronbeats,

I have updated the webflasher just now.
Now it is possible to load "optional" engines like TR707/TR909, that are sample based and require a lot of flash memory. 
You can also load the DX7 Sysex via WebFlasher.

Check out this website for D7 sysex files: http://dxsysex.com/SYSEX_DX7/V/dx7-sysex-V.php

I have used [ vrc-1003_percussion sysex](http://dxsysex.com/SYSEX_DX7/V/vrc-102%20wind%20instrument%20group%20(a).syx) so far without any issues (also upload via audio)

Actually, the popup "loading patch..." comes after everything has been successfully transferred. Can't explain what's happening - could you please try to load the SYSEX with the WebFlasher, and report back to me if it works via audio afterwards?

https://eh2k.github.io/□●/flash/

[image]

daltronbeats, 2023-06-10:
I was able to load the dx7 factory bank using the WebFlasher. 32 patches selectable and playable via the ui. the Pitch scaling is wrong though, on both Midi and CV in... 
still great for percussion stuff, cheers!

(I was also just about to ask for hold to octave shift and its already there!! nice work!)

eh2k, 2023-07-10:
HI @daltronbeats 
thanks for the Hint - i have fixed the wrong pitch scaling in the latest version.

(Experimenting with stereo and the rate parameter at the moment, Polyphonic Mode is also still on todo.)

Regarding "octave shift" while holding the encoder - I don't really understand the advantage - do you mean in general or specifically for DxFM Engine. If general, feel free to create a new issue.

eh2k, 2023-12-01:
@snakesystems fyi

themangoest, 2024-07-20:
What do you need to do for the plaits in Squares and circles?

eh2k, 2024-07-21:
@themangoest - all m-osc engines, except waveforms (braids) are basically plaits. All plaits drums are also available under `DRUMS`. The `SYNTH/ClassicVAVCF` is from the latest plaits latest update.

Plaits has a bunch of engines, depending on parameter ranges they are mixed or mapped simultaneously to out and aux on the original device. My goal was to separate them as standalone engines, and where are some engines that are not yet in squares-and-circles.