eh2k/squares-and-circles #45: Feature request - Insert levels for all machines on #4 not just FX

daltronbeats, 2023-07-16:
Hi again! Would be great if you were able to implement the  insert 1,2 & 3 levels on machine #4 regardless of if it's an effect or not.
At the moment I am using 3 parts of drums and have to add an effect machine on channel 4 to be able to get a mixed output on D (4). Would be great if I could add a 4th drum part and have them all mixed to the D output. Saves having to use an external mixer. 
Let me know if that would be possible.
Thanks

eh2k, 2023-07-18:
Hello @daltronbeats  - thank you for your feedback. I have already thought about a similar use case. One possibility (my favourite) would be to have two stereo outputs in general - (a+b) and (c+d): I.e. each engine/slot would have a pan + FX send parameter in the IO settings in addition to the already existing gain (effects could be selected similarly to quantizer). Effects could not be modulated in this mode  - unless you use them as an engine and occupy a slot with them. 

However - as I read you correctly - you would like to mix stereo to mono? That would be quite a downgrade for effects like reverb. If I misunderstand, just correct me.

daltronbeats, 2023-07-18:
This sounds like a good solution, I didn't think 4 drum/synth engines + FX would be possible. 
I would be happy with 4 engines mixed to stereo (or mono) so anything beyond that would be great. 

I'm just trying to avoid using 4 cables out to an external mixer, the FX for me are not critical

eh2k, 2023-07-28:
@daltronbeats check out the latest update. For more Info see  https://github.com/eh2k/squares-and-circles#mix--outputs or/and https://github.com/eh2k/squares-and-circles/issues/6#issuecomment-1656146698

daltronbeats, 2023-08-01:
Thank you! Will test this week

closed 2024-01-14 by eh2k