eh2k/squares-and-circles #47: USB Host

daltronbeats, 2023-08-09:
I noticed a mention of your interest in USB host in a previous comment. Was this ever implemented? I'd love to make a dedicated midi expander with seperate pots for each machine.

(Also looking forward to details on the FV1 implementation! I'm going to have to start making my 4th squares and circles soon 😁)

eh2k, 2023-08-11:
@daltronbeats,

i'm not sure i understand the question correctly. By USB host, do you mean USB-midi support?

Regarding Midi Expander.
My latest findings are that the following user interface would be quite practical:

[image]

Regarding FV1 .... more info when I am ready.

daltronbeats, 2023-08-12:
This looks great! Are the BD1, BD2 etc extra gate inputs?for additional engines beyond 1-4? 
Would be happy to beta test any new hardware when you are ready! That is my expertise..

Sorry I wasn't clear. The teensy 4.0 has 2 pins on the rear for USB host support. If I connect a class compliant usb midi controller to those would it work? (I'm already using the teensy main USB as a midi interface for triggering notes from a PC)

I understand I could send MIDI CC from the PC but it would be nice to have a small midi controller module next to my squares and circles module OR be able to connect a USB midi controller when I am away from the PC

eh2k, 2023-08-17:
I haven't done anything with the USB host on the Teensy yet. However, I have experimented with USB host on the DaisySM.

In general, I would probably prefer the serial/midi interface for an external controller. I had already thought about using the pins on the back for the Midi-Expander (you would have to solder something like this https://mou.sr/45eRDG8 to the teensy).

Regarding the second prototype:
The BD1, BD2 would actually just be buttons (with LEDs, like https://www.thonk.co.uk/shop/illuminated-tact-switches/). They are primarily used to select the track or to visualize the selection or triggs (these could also be named differently). The other two buttons (left + right) would be for navigation between the paramter page and io page - or extra pages like mixing. 1-8 would be CV inputs that also serve as triggers. A,B could be audio inputs, for example. C,D are the audio outputs. The current firmware concept would actually fit well with this. With only two audio inputs and mixing, 6 tracks would also be conceivable. At the moment there are 4 tracks/engines, as it fits well with the 4 DAC outputs.