eh2k/squares-and-circles #48: issues with DAC calibration

lancerotti, 2023-08-10:
Hi,

I was having some issues with the pitch in the oscillator engines so I decided to go through calibration once more. I'd done it when I had the  O_c firmware on it and everything seemed fine at the time.

Now I don't seem to be able to get any less than 2.48V out of the DAC channels when trying to calibrate at 0V.

I have 2 other O_c that I could try the same teensy on and see if the problem persists.

Any idea otherwise?

Thank you in advance
Nicola

eh2k, 2023-08-11:
Is it possible that in the case described only the Teensy is powered via USB and the module is not fully powered. I think the DACs need the full -12..12V to function properly.

lancerotti, 2023-08-11:
Possibly. It's a teensy that I didn't cut the trace of, so it was in fact just powered via USB.
But anyway, once calibrated on full -12 +12V it works perfectly
Thanks
N.

closed 2023-08-11 by eh2k