eh2k/squares-and-circles #53: Prototype

snakesystems, 2023-09-04:
Hi there. I’m looking forward to building a new uOC to run squares and circles but wondered if there was any movement on the development of the new prototype/version?

eh2k, 2023-09-05:
Hello,

thank you for asking.

I'll have to take the time to upload a demo to youtube sometime.

Generally it works as thought - the same software also running on the Daisy-SM - however I would make some hardware improvements on the second PCB batch.

 * All resistors on backside (if possible 805 size).
 * 74HC165 SO-16 instead of through-hole
 * Either way Midi-In based on BAT54S (currently there are jumpers for midi a or b)
 * Using KS01-BLV for tact switches
 * Using SPI-MOSI pin for receiving from 74HC165

Regarding encoders, I would try to use 24step without detent in the next build and test how it feels.

Generally I'm a bit tired of modules with two encoders and a display - so I'm also thinking if a bigger module would be better. 

But, I could imagine to make a second PCB batch, maybe already with pre-assembled parts.

Please send me a non-binding e-mail - with the following details and which option you would prefer:
 * Your location (Germany, EU, Worldwide)
 * A - Only Gerber files (private build)
 * B - PCB/PANEL only, 
 * C - PCB with all SMD components pre-assembled, 
 * D - PCB with all SMD components pre-assembled + two encoder + tact switches + through hole optocoupler IC
 * E  - You are a manufacturer / store and want to distribute modules to third parties ?!

In any case, you would have to source the display and the daisy-SM yourself.

snakesystems, 2023-09-05:
Ah that's great news and would love to see a demo video. I'll go ahead with
the OC based version anyway but would be very interested in the new
version.
I'm in the UK and would want version B for a self build or C happy to be
added to the list if you needed test builders.
My whole case is DIY including Temps Utils, OC, Befaco, Jakplug MI clones
and Braids among others.

Ric

eh2k, 2023-09-24:
I finally managed to make a video. Since the DSM has two audio inputs, you can use it as a recorder in the background (5min into ram). Actually, it is an experiment, but somehow very useful.

https://youtu.be/lNC6JGtgr8k

snakesystems, 2023-09-25:
Looks and sounds great!

snakesystems, 2023-09-25:
Are you anywhere near needing test builders? Happy to pay for PCBs and
panels of course.