eh2k/squares-and-circles #55: U plan to share your hardware design?

stilltryin, 2023-10-28:


eh2k, 2023-10-29:
I have designed several RevB variants that have not yet been built. The first version as seen [here](https://user-images.githubusercontent.com/11786292/276744247-0f8a0b3b-1774-41f6-bb99-11d4f8b384d0.png) works as expected. Maybe I have one plain PCB+panel left (If you are interested in building it, just send me an email for details).

The RevB I would like to discuss in a group of RevA builders/users. And if there is a RevB, almost everything (daisy patch sm and other chips) from the RevA can be reused there.

I may end up publishing just the hardware design, and/or the PCBs may be available somewhere.

stilltryin, 2023-11-06:
Hey,

yeah cool would be interested to build one. I need a Teensy 4.0 if i got it
right? can make the pcbs here since i literally live where all the FAB
HOUSES are lol....think its cheaper then send a set....