eh2k/squares-and-circles #57: CV/V/OCT outputs a lower voltage

eh2k, 2023-11-19:
Hi @eh2k !
I have been fiddling around with S&C past versions (circa 2022) for few monthes, but found a strange issue with last version (508de0e) : the CV/V/OCT engine outputs a lower voltage (with an approximative ratio of 0.66 of what it should be) than the one written on the bottom of the screen. On the other hand, when I do the calibration, the output voltage is the same than the one displayed. I investigated on this and I guess something strange is happening in the "push" function of the machine binary, as I traced the cv value and it is coherent until that call. Once again, the hardware shouldn't be involved as the calibration process outputs the displayed voltage.
Am i missing something ?
Thanks again for the work
Best regards
j.

_Originally posted by @jrachedi in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-1817572010_

eh2k, 2023-11-19:
Hi @@jrachedi,

thanks for the hint.

As I remeber, I have also noticed something strange some time ago. 
   * Modulation did not work properly
   * Finetune was quantized.

I recently redesigned the CV engines, using the code of the internal modulations. The new scope view is quite nice.

Could you please try to reproduce the problem with the latest version [83cf6aa](https://github.com/eh2k/squares-and-circles/commit/83cf6aad241db43ff9277e9eb1d95e04bd680aad) and let me know if this is still the case.

Here are some additonal information:
  * The quantizer affects the "V_OCT " parameter.
  * Check the transpose in the IO page.
  * If another engine is mapped to the same output, possible the output is calculated to audio scale (probably the case?).

jrachedi, 2023-11-19:
The problem reproduces with 83cf6aa … sorry :)
About the points you mention:
- this happens with the quantizer disabled
- this is not a transposition issue as the intervals are out of tune on all the range. When I control with MIDI and play Cs in octaves, the display screen displays a 1v difference, my voltmeter displays a 0.66v difference, and if I control a VCO, this is nothing sounding like an octave. If I do it without MIDI just turning the little keyboard encoder control, the half steps sound smaller than a real half step and if I go from C4 to C5 this is the same strange 0.66 ratio interval. So MIDI is not involved.
- the CV/V/OCT engine is the only engine running on the module

jrachedi, 2023-11-19:
It reproduces with 27df093 as well. 
The scope view is indeed a pretty neat feature !

eh2k, 2023-11-19:
Just pushd https://github.com/eh2k/squares-and-circles/commit/7d07b9af2e47948ba3e6322deb6385497a935fb6.
This issue should be fixed there.
Each engine has an internal volume parameter (IO-page). This is not visible in case of CV engines and should now always be 1.0.
Could you please check if it works for you. Thanks for your feedback.

jrachedi, 2023-11-19:
Your last commit fixed my issue !
Thanks a lot
Julien

koswir, 2023-11-25:
> Just pushd [7d07b9a](https://github.com/eh2k/squares-and-circles/commit/7d07b9af2e47948ba3e6322deb6385497a935fb6). This issue should be fixed there. Each engine has an internal volume parameter (IO-page). This is not visible in case of CV engines and should now always be 1.0. Could you please check if it works for you. Thanks for your feedback.

Hey!
First of all I love S&Q! eh2k thanks for this amazeing firmware!
I was on my way to share this bug too, but jrachedi was faster! :)

7d07b9a finally works here and I'm so happy about that but I've noticed another problem.
The Slew parameter works right up to about 72/73. If I set it to a higher value - the glide is longer but never reaches the desired pitch.
Please check that out!
Koswir.

eh2k, 2023-11-26:
@koswir -Thanks for your feedback, I'm glad you're having fun with the firmware. 

The V/Oct engine may be inconspicuous, but together with the modulation you can do some crazy things with it. It could well be that some usecases arise that I didn't even think of during development. If you like to describe how you use it in more detail, just post them here - maybe I can use the information to include it as documentation in the wiki...

Regarding the Slew parameter. It is actually a simple ONE_POLE filter: ` ONE_POLE(cv, cv0, powf(1 - slew, 10));` where `slew` is a value from 0 to 0.5. I'll have to take a close look at this, it may well be that the range from 0 to 0.28 would be better. Then the maximum turn to the right would correspond about to the current 72/73.

koswir, 2023-11-26:
> Regarding the Slew parameter. It is actually a simple ONE_POLE filter: ` ONE_POLE(cv, cv0, powf(1 - slew, 10));` where `slew` is a value from 0 to 0.5. I'll have to take a close look at this, it may [...]

oh no please do not limit the range! Glide at 72 is really very short.
My workaround for longer slew rates is to set 72 than add an envelope to push the value up but only for a desired time. :)
Thanks!
Koswir

closed 2024-01-14 by eh2k