eh2k/squares-and-circles #58: FV1emu

koswir, 2023-11-25:
Hey.
I am a new user of the amazeing S&C firmware.
I have noticed that FV1emu button at the flasher screen. 
I was searching your project page and forum but faild to find an answer how to use that and what kind of file I could load.
Sorry if I am missing something obvious.
[image]
Thanks!
Koswir

eh2k, 2023-11-26:
This allows FV1 ROMs to be loaded as engines.
At the moment I still have a few points on the todo list, and the whole thing is not finished yet. 

You can find out what it is in general here:

https://github.com/eh2k/vcvrack-fv1-emu

koswir, 2023-11-26:
Impressive list of fx!
Can't wait to try the finished thing!

koswir, 2024-01-30:
@eh2k 
Hey, could you provide an approximate timeframe for when it will be possible to load the effects?
I cant wait to try that. :)

eh2k, 2024-01-31:
hi @koswir,

here is a webpage from which you can compile and upload SPN files as "S&C Engines". 

[https://eh2k.github.io/□●/fx.html](https://eh2k.github.io/%E2%96%A1%E2%97%8F/fx.html)

I haven't actually followed this for a while. Because I'm not really happy with the execution performance and the "sound". And it's damn time-consuming to test and since the OC has no real audio inputs, it is actually less interesting. (Perhaps the S&C-DSM0 is more interesting here)

However, everything I've done here was really only for experimental purposes. Looking back, I would probably prefer faust effects nowadays (see https://github.com/eh2k/squares-and-circles/blob/main/app/FX/Rev-Dattorro.cpp).

The Web-FX compiler runs on my private web server - i.e. whenever you click on the Compile&Flash button, a server process starts that compiles an  ARM thumb executable from the specified SPN file. 
Like the TR707 SampleRoms, this is then flashed to a separate memory area on the Teensy and can then be started later as an S&C engine.

I don't really want to run this publicly - so I'll probably take it offline again after a while. 

You can only compile SPN files that are on github. I will track them and maybe take a closer look if I find the time. 
With all the FV1 effects that you can find on github, it's generally a bit unclear regarding the licence. 

If you have any questions or maybe an fv1-fx that you like to share, just post here...

koswir, 2024-02-02:
@eh2k
Thank you so much :)
I have tried to flash from here https://mstratman.github.io/fv1-programs/ but I've always got "500" message.
[image]
Sorry if I am missing something obvious.

eh2k, 2024-02-02:
Hi @koswir,

The upper url is a github.io "pages" domain - I handle it like as an invalid website at the moment.

Only direkt "github" urls can be used, like https://github.com/mstratman/fv1-programs/blob/master/public/files/spinsemi/dance_ir_h_l.spn

All files from here should work:

https://github.com/mstratman/fv1-programs/tree/master/public/files

koswir, 2024-02-02:
@eh2k 
Thank you! It works great!
Some effects sound very good (eg. slocum-phaser.spn and softclipping_overdrive.spn)
I am going to check them all later and raport here :)

> I don't really intend to run this publicly in the long term - so I'll probably take it offline again after a while.
Any way to save or backup the compiled file?

koswir, 2024-02-02:
@eh2k 

> If you (or someone who reads this) would like to help me select good effects, a JSON file with the following content would be the way. Which you could also maintain in a own repo.

I am going to test these effects and pick up the good or at least useable ones.

koswir, 2024-02-04:
@eh2k 

> As indicated, my concerns are less technically, the point is I don't want effects with an unclear licence to be part of the firmware.

Will you have any doubts about the licenses if I create effects specifically for S&C in SpinCAD-Designer?

eh2k, 2024-02-05:
@koswir 

I have redesigned the fx-compiler page so you can now load a Json file from local computer and flash several SPN files in one go.

The naming must therefore be done in the JSON file.

Regarding Spin-Cad - I don't know how it is exactly - as far as I know the building blocks were mostly copied from the SpinSemi forums. To me the effects were often just combinations of chorus + reverb + filter. But there were definitely a few interesting ones.

I think you have to use the effects for a while to tell if they are really good. Often together with a sound source (e.g. I like the combination of TR707 samples + Gated-Reverb).

If you experience crashes, please report it. These happen to me from time to time and I haven't figured out what the cause is yet.

koswir, 2024-02-06:
@eh2k 
aw great! :)
I have made some fx in the SpinCad today.
https://github.com/koswir/FV1-programs-by-koswir
Sadly I can not load them thru .json. Have you limited an url from where spns can be installed?

koswir, 2024-02-06:
@eh2k
Hey, I have tested these fx I've made and there is some kind of bug I can reproduce.
When I select a knob's modulation for the first time after loading on one of my fxs, it crashes immediately when selecting a modulation source.
On plugins you have built in it works without a problem and also, if I select a modulation source on one of them then switch to one of my fxs - it works fine.
[image]

eh2k, 2024-02-07:
Hi @koswir,
thanks for the crash report.
I was actually able to reproduce the crash.
In my case, the problem occurred with the "Gated-Reverb" when a modulation was set or another engine was loaded in the same slot.

I have just pushed a new commit. It would be great if you could upgrade to the latest version and use it from now on.
I have cherry-picked some changes from my dev branch there. Especially the FV1 engines now only need about half of the RAM - because I reduced the delay buffer to 16 bit.

You would have to transfer or compile the effects again, after that the crash should no longer occur. 
Otherwise just post a picture of the crash report with a short step-by-step description.

eh2k, 2024-02-07:
@koswir,

I am writing this post here that I will delete immediately, and hope you read the email that will be sent.

I have noticed that you have forked my "eh2k.github.io" repo. The fact is that I only use the repo only because of the github.io domain, and it actually accesses my private server in the background.

In short - the repo only contains a small part of the whole (some quick and dirty javascripts with a lot of copy&paste), and only works when running on eh2k.github.io.

Could you please delete this repo - I don't want it to somehow spread on github.

It's ok if you clone the code locally, and maybe try to understand it if it makes any sense to you.

If you have any questions you can also simply contact me by email.

koswir, 2024-02-09:
@eh2k 
Thank you! It works now!

koswir, 2024-02-09:
I'm going to delete this message in a second. 
I've tried to contact you at eh2k@github.com but got Delivery Status Notification (Failure). 
I have already removed that fork, sorry.
Please mail me at kosmiczny@wir.pl

koswir, 2024-03-17:
It took me some time, but I've gone through several effects from the provided link: https://github.com/mstratman/fv1-programs/tree/master/public/files

Since I couldn't check them all, I'll share feedback on the ones I did test.

@eh2k, as you mentioned, it's tricky to identify the authors. Some files have links to topics on the Spin Semiconductor forum, while others include credits (for example, RevRev.spn was made by Malekko).

**In my opinion, every S&C user should load them at their own risk.**
Therefore, I've chosen not to provide a JSON file with the tested ones. Someone may create it if needed.

The ones that function perfectly are marked in bold. Those without comments haven't been tested.

**8_seconds_delay.spn** - works

_ARPEGGIO.spn
ARPEGGIO_MINOR.spn
ONIAN.spn_

**RevRev.spn** - works (a very, very cool reverser from Malekko)

_afx_reverse_delay.spn - loads but no wet signal_

_aliaser.spn - it loads but I think its kinda broken_

**bm-pi.spn BM Pi** - works great (Distortion, Controls: Gain, Tone)

**chorus-dual-rate.spn** - works great (Controls: Chorus level, Rate 1, Rate 2)
 
_crusher.spn - works but controls do not work_

**dattorro.spn - works, (it's a reverb but the one already included is better anyway)**
 
**faux-phaser-2.spn** - works (Controls: Speed and Width, Time, Level and Feedback)
 
**faux-phaser.spn** - works (Controls: Speed and Width, Time, Level and Feedback)
 
_freeverb.spn  -  kinda works but it's buggy - which is sad because it may be the longest reverb from this base_
 
_greenwood-delay.spn_
 
_hall_goldmine.spn - loads but is sounds like a dry signal_ 
 
_improved_octave_up.spn - kinda works but sounds weird_
 
**interesting-1.spn** - works (it's a mixed mod/reverb/delay fx)
POT0 is mostly flange speed
POT1 is reverb time combined with a swept high pass on a parallel delay line
POT2 is the mix level for the flange and delay. Reverb comes through in any case.
 
**interesting-2.spn** (overdrive version)  -  works 
POT0 is mostly flange speed
POT1 is reverb time combined with a swept high pass on a parallel delay line
POT2 is the mix level for the flange and delay. Reverb comes through in any case.
 
**lfsr_random.spn** - works but it's just a random noise source
 
**modulated-echo.spn** - works (Controls: Time, Depth, Level)
 
_multipitch.spn_
 
_parametric-eq-7.spn - loads but is sounds like a dry signal_
 
_pingpong.spn_
 
**pitch-transpose-stereo.spn** - works great (a stereo detune/chorus, Controls: L channel Pitch, R channel Pitch, Delay? (I'm not sure does it work))
 
**resonator.spn** - works (Controls: Frequency, Resonance (inverted))
 
**rev_pl_1.spn** - kinda works but meh
 
_rev_pl_2.spn_
 
_rev_pl_3.spn_
 
_reverse_delay.spn - loads but no wet signal_
 
_sawtooth.spn_
 
**shimmer-1.spn** - works
but already included revebs are better
 
**shimmer-2.spn** - works
but already included revebs are better
 
**shimmer_drAlx.spn** - works (but it doesn't sound as described - eg. in the description POT0 = Reverb time (0 to 10 seconds) but in my opinion reverb time is max 2 seconds, Controls: Reverb time, Treble, Shimmer)
 
**slocum-phaser.spn** - works great (Controls: Phase rate (manual if fully CCW), Sweep width (or manual), Num of Stages (4,6,8,10))
 
_slowgear-2.spn_
 
_slowgear-simple.spn_
 
**softclipping_overdrive.spn** - works great (Controls: Gain Threshold, Volume, Tone (inverted))
 
_spring_verb.spn_
 
_stereo-dual-flange-multi-tap-delay.spn_
 
**stereo-ring-modulators-with-chorus.spn** - works great (Controls: Blend, Carrier offset, Chorus)
 
_stereo-tremolo.spn_
 
_thru-gain.spn_
 
**tremolo-shapes-reduced.spn** - works (Controls: Wave shape, Depth, Speed)
 
_tremolo-shapes.spn_
 
**triple-delay-feedback.spn** - works great (Controls: Time 1, Times 2 & 3, Feedback)
 
**up-down-octaver.spn** - works (but it's one oct up and two octs up (no down oct as name suggests)
 
**whammy.spn** - works (Controls: Whammy, Pitch range: DIVEBOMB, OCTDOWN, FIFTH, OCTAVEUP)

koswir, 2024-03-18:
@eh2k Hej have you disabled the fx compiler?
Seems like it doesn't work :(

eh2k, 2024-03-18:
@koswir - thanks for the review. I will also test/listen to the effects and then maybe list them somewhere.
Please try the Fx compiler again - or press CTRL+F5 in the browser to refresh the browser cache.

koswir, 2024-03-19:
@eh2k the plugins I've made with SpinCAD flash without a problem.
For some reason I can not flash a .json from https://github.com/mstratman/fv1-programs/tree/master/public/files
[image]
```
[
        {
        "name": "FX/BM Pi",
        "pot0": "Gain",
        "pot1": "Tone",
        "pot2": "",
        "url": "https://github.com/mstratman/fv1-programs/blob/master/public/files/bm-pi.spn"    
    }
]
```

eh2k, 2024-03-20:
@koswir - please try replacing the blank in the name "FX/BM Pi" to "FX/BM-Pi".

koswir, 2024-03-20:
> @koswir - please try replacing the blank in the name "FX/BM Pi" to "FX/BM-Pi".

Thank you!

@eh2k btw is there a way to skip a pot if a particular fx uses less than three?
When I leave it blank - s&c creates a knob with a default name.

rootapprox, 2024-11-07:
hello guys! can you please tell me how to Load JSON files to the Fv1 from the effects that you suggested ? when i download i have the files in .spn format ..

eh2k, 2024-11-07:
@rootapprox 

You would have to create a json file according to this [template](https://github.com/eh2k/squares-and-circles/blob/main/lib/fv1/factory.json) where you define the FX and the paramter names and upload it your github repo. The referenced SPN files must also be provided as URLs in the json file and be publicly accessible.

You can then enter your json-url on the fx page and compile&flash the effects directly to your device.
The effects are copiled and stored on my server. All experimental...

[https://eh2k.github.io/□●/fx.html](https://eh2k.github.io/%E2%96%A1%E2%97%8F/fx.html)

koswir, 2024-11-07:
> hello guys! can you please tell me how to Load JSON files to the Fv1 from the effects that you suggested ? when i download i have the files in .spn format ..

I can just share the files I use 😸 
Here are my favourites from the list:
[fv1-programs-set1.json](https://github.com/user-attachments/files/17670742/fv1-programs-set1.json)
[fv1-programs-set2.json](https://github.com/user-attachments/files/17670743/fv1-programs-set2.json)

And if you want to try the ones I made:
[koswirfx2.json](https://github.com/user-attachments/files/17670745/koswirfx2.json)
[koswirfx1.json](https://github.com/user-attachments/files/17670746/koswirfx1.json)

rootapprox, 2024-11-08:
Oh Thanks @koswir !! Ill be interested to try more if you got in the future ! Thanks again!!

eh2k, 2024-11-08:
@koswir 

fyi - I have edited the links in your last post, so you have only 1-click to load it. Yesterday I modified the FX website - you need to  have a public FX-JSON-URL as url parameter, or just paste the link on the page and press load. Consequently, you could also put the JSON in your repo.

`https://eh2k.github.io/□●/fx.html?json=<fx-json-url>`

koswir, 2024-11-08:
> @koswir
> 
> fyi - I have edited the links in your last post, so you have only 1-click to load it. Yesterday I modified the FX website - you need to have a public FX-JSON-URL as url parameter, or just paste the link on the page and press load. Consequently, you could also put the JSON together with the SPN files in your repo.
> 
> `https://eh2k.github.io/□●/fx.html?json=<fx-json-url>`

That is super dope! :D thanks

eh2k, 2026-10-31:
Closing this old question: since the 22026-10-31 - 3e930ee the web app has an FV1emu dialog.
There are no json banks anymore - add directly spn or bin files and name the parameters in the ui...

Effects can be played in the emulator before [Flash to module]. 
The LOG/EXP instructions are also more exact now. For anything new, please open a fresh issue.

closed 2026-10-31 by eh2k