eh2k/squares-and-circles #65: Black screen after updating firmware to 690cfd3

koswir, 2024-01-23:
Hey. Thank you for an another great update - the knob visualisation feature is dope!

Sadly two of my three o_c's can not update to 690cfd3.
The one that updated succesfully had a black screen at first too but somehow after a reset it worked.

With other two devices I tried everything I could think of:
update from a Mac, from a PC, different usb cable, update to 2023_0ecb86e, different usb port.

If the update is done there is a message "finished" but the screen remains black  no matter I do.
When I connect o_c after update to the flasher it loads the page with settings and other engines.

I already installed a few updates on these devices without problems. This is first time I have a problem.

To check are they ok, I have installed Phaserville suite firmware on both devices - and they work fine.
Then again I tried to install s&c - same black screen.

Please help! I want to tweak the hell out of the new firmware :)

eh2k, 2024-01-23:
@koswir 

Thanks for reporting. Let's narrow down the problem - I hope we can find the cause quickly.

Can you please answer the following questions:
  * After flashing the firmware version 2023_0ecb86e version - then everything works - correct?
  * After flashing the firmware version "main_690cfd3 (latest)" the display stays black.
     *  In this case I need to know if the firmware is running in the "background". 
     * Could you please tell me if in this case the O_C is recognised in the webflasher or if it says "No midi device detected." To be sure, press F5 in the browser a few times. If the "Upgrade" button appears, the firmware is running an we have only a display issue.

koswir, 2024-01-23:
> * After flashing the firmware version `2023_0ecb86e version` - then everything works - correct?
No, I couldnt flash it too.

> * After flashing the firmware version ~`main_690cfd3 (latest)`~ `main_1c37ee7 (latest)` the display stays black.
Yes. I tried the 1c37ee7 and the display is still black.

> * In this case I need to know if the firmware is running in the "background".
>   * Could you please tell me if in this case the O_C is recognised in the webflasher or if it says "No midi device [...]
yes, it probably runs in background - that's what I ment by "When I connect o_c after update to the flasher it loads the page with settings and other engines." 
It doesn't say "No midi device detected."

The weird thing that happend in meanwhile - one of these with main_690cfd3 sometimes starts - once per maybe 20 turn ons.
I do not think its a problem with o_c hardware - if I flash Phaserville - it works every time. Also there is plenty of power in the box.

Could I try to load an older version? previously I have got one of these 2023-11-19/ 2023-12-18/ 2023-12-03.

koswir, 2024-01-23:
New problem. 
The third o_c which was fine so far (with 690cfd3) has stopped working after I have pressed right encoder and right button (reset).
Now I can not turn it on.

eh2k, 2024-01-23:
OK - thanks for testing out - lets try the 2023-12-18 version using this link:
 * https://eh2k.github.io/%E2%96%A1%E2%97%8F/flash/?ver=firmware_OC_T40_e7f2df2

There are a few minor changes to the display code in the "moving knobs" version - however, version 2023_0ecb86e should not be affected by this. 

I'll have to check again tomorrow, you mentioned Phaserville - it could also be that the settings in the eeprom memory are being overwritten somehow.

eh2k, 2024-01-24:
just pushed `8278f73` - possibly the CRC of the eeprom settings check failed, resulting in uninitialized settings variables (should be fixed now) - this could explain the behaviour you describe. 

If the webflasher recognizes S&C Midi, please try switching the DISPLAY_SP1/SP2 or/and ENTER_CALIBRATION. Here the eeprom is written with valid values.

[image]

koswir, 2024-01-24:
> OK - thanks for testing out - lets try the 2023-12-18 version using this link:
>
> * https://eh2k.github.io/%E2%96%A1%E2%97%8F/flash/?ver=firmware_OC_T40_e7f2df2
>
> There are a few minor changes to the [...]

This version works perfectly on every device. Thank you. 

Could you please make some kind of backup for older version?
Something like a downloadable flasher or maybe even a hex?
I am so in love in Squares&Circles that I have removed all other vcos... and a situation like this disables my system completely :)

> just pushed 8278f73 - possibly the CRC of the eeprom settings check failed, resulting in uninitialized settings variables (should be fixed now) - this could explain the behaviour you describe.
>
> If [...]

Sorry to say that but this version doesn't work either.
The webflasher recognizes S&C Midi but pressing Display or Calibration do not help.

Maybe the problem appeared eariler but I havent flashed that late 2023 builds because I assumed they are DSM0 updates.
I will keep 2 of my o_cs downgraded for now.
The third one is going to be the betatester :)

eh2k, 2024-01-24:
Thanks for the quick response.

Could you please help me to narrow down the problem even further and try the version from 2023-12-21:

https://eh2k.github.io/%E2%96%A1%E2%97%8F/flash/?ver=firmware_OC_T40_5b0ac34

This is the version immediately after the 2023-12-18 (that works) and `2023_0ecb86e` version (that is not working).

With version `2023_0ecb86e`, I actually intended to provide a fallback version because I received some feedback for it.
If you confirm that `5b0ac34 `works, I could provide it as the stable 2023 version.

Precise Feedback is very important in such projects. If people simply switch to older version because it is easier as reporting an issue, it's bad for everyone.

I assume you have the "blue OLED" ?

koswir, 2024-01-24:
Thanks for the quick response.

> Could you please help me to narrow down the problem even further and try the version from 2023-12-21:
> https://eh2k.github.io/%E2%96%A1%E2%97%8F/flash/?ver=firmware_OC_T40_5b0ac34
> This is the version [...]

It works seamlessly!

> Precise Feedback is very important in such projects. If people simply switch to older version because it is easier as reporting an issue, it's bad for everyone.

Yes, of course precise feedback is important. From now on I will check every version and provide the best feedback I can.

Considering it from the user's standpoint, our goal is to create music. We can't predict the future, and if, for some reason, you abandon the project and the only version available isn't functional, the whole projekt will be lost.
Please, at least consider leaving more older versions accessible in the web flasher.

> I assume you have the "blue OLED" ?

No - all of them are white?
Why do you ask?
[image]

eh2k, 2024-01-24:
HI - thanks again for your patience.

The good news is, that I have now been able to reproduce the issue on a breadboard. I hope that my last push `main_2ad0024` is solving it. For whatever reason I had to reduce the SPI speed during initialization - after that the display on the breadboard started working.

So far, only the blue OLEDs have caused problems, which is why I had this assumption.

Just a thought - I can understand the idea of offering all versions. Generally, I would do this in a closed area - but I have no idea how to decide who has access or not.

n0ahg, 2024-01-24:
Hi,

As an extra data point, I tried to update my white screened O_C earlier
today and was met with a black screen.  This newest update appears to fix
the screen issue for me.

Thanks for the hard work
Ray

koswir, 2024-01-24:
Yay!
> The good news is, that I have now been able to reproduce the issue on a breadboard. I hope that my last push `main_2ad0024` is solving it. For whatever reason I had to reduce the SPI speed during [...]

It works! Thanks!

> Just a thought - I can understand the idea of offering all versions. Generally, I would do this in a closed area - but I have no idea how to decide who has access or not.

Man what you do is amazeing! Thank you for the hard work.
Patreon or a donation would be fair!

closed 2024-04-10 by eh2k