eh2k/squares-and-circles #66: Connection from Webflasher, not always.

jpnielsen, 2024-02-08:
Case image 1:
[image]
[image]

Connection to teensy seems OK 

case image 2:
my audio interface is turned on, and the webmidi tester is reloaded:
[image]

refreshing Webflasher page doesn't show the options&engines page:
[image]

jpnielsen, 2024-02-08:
I think i got it working,

After running reinstalling my Presonus Studio 192 driver and doing "sfc /scannow" in windows 11, the medbmidi tool writes this:

[image]

Now, "Squares&Circles" has the same number for input(0) and output port (0).

jpnielsen, 2024-02-08:
after a power reset of the eurorack modules, I'm back to the same as before.

No  webflasher options/engines.

Webmidi tester shows different in/out numbering.

[image]

Thank you.

jpnielsen, 2024-02-08:
Just noting that other devices have different names for IN and OUT.

[image]

eh2k, 2024-02-09:
@jpnielsen,

thank you for the detailed report.
I suspect there is some issue in my javascript code. I've gone through some cases and added a few error handlings.

As you can see in your first posting, the second screenshot does not show "no midi device detected". This means that "Squares&Circles" was found in the WebMidi devices. I suspect it could be a timeout in further communication. Could you please open the Developer Console in the browser (press F12) and see if red error messages are displayed there?

[image]

jpnielsen, 2024-02-10:
Yes, a timeout, when the Presonus Studio 192 is online.

[image]

Power off the Presonus Studio 192, and webflasher is OK:
[image]

Pepebotika1980, 2024-02-10:
Is it possible to download the firmware .hex file? I can't get webflasher to work for me. I have tried different computers, cables, USB connections and I always get the same error: ReferenceError: delay is not defined- please try again. On the other hand, if I load the phazerville firmware with teensy.exe I have no problems. thank you.

eh2k, 2024-02-10:
@jpnielsen - I have just updated WebMidi.js from 1.3.5 to 1.3.8 - maybe this will solve the issue.

I think I will introduce a MessagePopup in the firmware with the next update, so that you can safely track and verify whether the commands from the PC arrive at the Teensy, or the response is lost for some reason.

@Pepebotika1980 - Possible this information will help you in some way https://github.com/eh2k/squares-and-circles/issues/43#issuecomment-1646617719

Otherwise, it would help me if you could provide more details, such as which browser and operating system you are using. Screenshots of the error message would also help.

jpnielsen, 2024-02-10:
Unfotuunately, I see no change. I can still only connect to the daisy firmware from the webflasher, when i turn off my Presonus Studio 192.

I don't think i'm going to use the midi on the presonus - but it doesn't help when i turn it off in the device manager, and the available settings in the Presonus Universal control don't seem to make a difference.

[image]

I was thinking you maybe could give us the possibility to change the name for the input and output midi devices - that would be usefull too if we have more O-C/s connected.

I would probably call mine "OC1 IN", "OC1 OUT", "OC2 IN", "OC2 OUT", "OC3 IN", "OC3 OUT", "OC4 IN", "OC4 OUT" and "OC4 IN", "OC4 OUT".

Or do we have some other unique identifier going on that i'm not aware of ?

Pepebotika1980, 2024-02-10:
Hello eh2k, I have tried to update the module again and it still gives the same error, I am attaching some screenshots, I am not sure what you need to see, if not, tell me what exactly you would like to see.
[image]
[image]

I am using this version of Chrome in windows 10:
[image]

I have also tried to load that firmware that you told me about and it hasn't worked for me either. When I load it, the module waits for an update, the webflasher detects it, but when I have to click on "flash", it returns the same error.

[image]

[image]

That's all the information I can give you. . Thank you so much

eh2k, 2024-02-10:
@jpnielsen - I took a close look at the webmidi tester - Thanks for that, we can actually use this for debugging
   - what exactly do you mean by "daisy firmware" - do you see the uO_C Image?

@Pepebotika1980 thanks for the infos - I have fixed a small bug (which actually only occurred after flashing a bare-teensy). 

I hope we're not mixing two different issues here - but I think it could be the same issue: 
The S&C & WebMidi somehow doesn't work poperly in some rare cases.

Before I make any further adjustments, we have to ensure the following cases are working.

1. Makes sure an S&C version is flashed, or flashes this [S&C-Loader hex file with the TeensyLoader](https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/squares-and-circles-teensy4XX-loader.hex)

Goto https://studiocode.dev/webmidi-tester/ 

1. Verify `Squares&Circles` Midi device is detected - it looks like this to me [image]
2.  Copy&Paste this Code `0xF3 0x4C` into "Arbitrary bytes" and send
    - The command should open/close the Machines/Engines menu (see O_C display)
3. Copy&Paste this Code `0xF3 0x56` into "Arbitrary bytes" and send
    - Thes command asks for the flashed firmware version - you should see the response as 8x receive messages immediately after the send message [image]

Thank you for your patience.

jpnielsen, 2024-02-10:
> * what exactly do you mean by "daisy firmware" - do you see the uO_C Image?

corrected,  teensy firmware  offcause.

Pepebotika1980, 2024-02-10:
yeah!! I already have the module working correctly, thanks to you for your patience and for the great work you do!

eh2k, 2024-02-10:
@jpnielsen 

In addition to the O_C, the webflasher supports another module based on the Daisy Patch Submodule which a few people have. 
This is all recognized automatically by the device-id and you only have one click to flash.

But back to the issue. Let me summarize briefly:

The issue only only occurs when the "Presonus Studio 192" is turned on. 
When it's off, everything works fine (Please confirm).

Regarding the timeout - it occurs if there is no response 5 seconds after sending. I don't think it makes sense increasing it.
I don't know what else is running in the background on your PC. For example, you can run the WebMIDI Tester in parallel in a second browser window; it receives the same MIDI messages. It could be that something like that is causing the timeout in the WebFlasher.

I think the "device names" are different since I'm on Linux and you're probably on Windows.

jpnielsen, 2024-02-11:
I tried several combination of midi devices now - the result is, that i can only get your WebFlasher config tool to work, If there are no other midi devices connected. It's not just "Presonus Studio 192" - also Behringer Neutron, Behringer Model D.

jpnielsen, 2024-02-11:
I wanted to reflash the latest firmware 

But the flasher finishes very fast, and nothing is flashed to the teensy.
I first "erased" the teensy by werwriting with squares-and-circles-teensy4XX-loader.hex using the teensy loader.exe from pjrc.com

eh2k, 2024-02-12:
Hi @jpnielsen,

Today I connected two teensys to a Windows PC and tested the Webflasher. I think I was able to reproduce the issue you described. It looked like I got into the state when I unplugged one of the teensys while the WebFlasher was open. It seems to me that WebMidi generally is not supporting multiple devices that have the same name. I also had the case that the browser didn't recognize any MIDI devices at all - after I closed it and reopened it, it worked again. 
I think it's safe in this a situation to unplug all teensys, restart the browser and plug them back only once at a time.

The next days (maybe tomorrow) I would introduce that every S&C device has a unique MIDI name, I hope that this will fix this issue.

jpnielsen, 2024-02-13:
> seems to me that WebMidi generally is not supporting multiple devices that have the same name

I have only connected one teensy so far. Unique S&C MIDI names sounds promising.

jpnielsen, 2024-02-13:
I have flashed the latest, looks good - and I can use it without disconnecting other MIDI devices.

[image]

jpnielsen, 2024-02-13:
I got out the champagne too early...

activated "707" from the webflasher - it uploads - and reboots the teensy after a while.

Now the numbering has changed:
[image]

and the webflasher tool cannot connect.

eh2k, 2024-02-13:
@jpnielsen 

I will continue to test this with multiple devices for a while and thinking about to add the option to select one S&C device if several are connected in the WebFlasher .
Multiple S&C devices are now listed by WebMidi on my Windows PC. If it doesn't work somehow, maybe restarting the browser helps.

[image]

jpnielsen, 2024-02-14:
After updating to the latest firmware (main_9cd30a1), and rebooting.
the teensy displays:
"04E9E512673C"
"Ready to update"

eh2k, 2024-02-14:
@jpnielsen 

thanks for testing and reporting.

I've found a minor issue after some testing and have just pushed another update.
Previously, usb-midi messages were only processed after the splash screen, which takes about 5 seconds on startup  - I think this was the cause for the timeout.

In the current Webflasher multiple devices are supported, you can select the midi device in the top right combo box.

If no Midi device is recognised, you propable have a bare teensy and have to flash the "Loader" first (I have added a short explanation in Webflasher today - hope that it is understandable so far). The reason for this is that I plan to support different Squares&Sircles devices and everything that happens after the loader will be the same for all devices. 

My goal is that it will only take one click to install an firmware update.

Regarding your last message:
I assume you have only flashed the loader, because the "Ready to Update" is displayed.

eh2k, 2024-02-15:
You can actually increase the flashing speed. I have recently reduced this because a few users have reported issues. 
The current setting is slower, but worked more stably so far.

If you want to have it faster, you can change the following parameters in the Webflasher URL (I am using the following parameters without issues)

`....&transmit_delay=0&transmit_frame=64`

eh2k, 2024-05-07:
Hi @OvErDrIvEwItHdIsTo 

thanks for reporting.

I have just tested the firmware flashing (even with french translater). It works for me. 
Maybe the server was temporarily unavailable or there is something wrong with the browser cache.

Please try the following:
  * Open Webflasher in your browser.
  * Press ‘CTRL+F5’ before the ‘Flash/Upgrade’.
  * And then perform the ‘Flash/Upgrade’.
    * If the error occurs again, try the same in browser incognito mode (or another browser/device)

OvErDrIvEwItHdIsTo, 2024-05-08:
Thanks, I will try it when come back home.

OvErDrIvEwItHdIsTo, 2024-05-09:
Hy, I try your solutions but it does not function, it's always the same message. 
So I install an old .hex firmware for using my module this week for a party, but I think it is the first firmware with not many functions and quality... 
Do you have another .hex files, even if it's not the last one ? 
Thanks for your time.

eh2k, 2024-05-09:
Hi @OvErDrIvEwItHdIsTo,

thanks for trying again and for your patience.

I think I have fixed a possible bug in webflasher. 
Could you please try it again and give me a step-by-step description of what you are doing.
  - Do you "flash/upgrade" an older version (Stage 2) ?
  - Or do you trying to install the "loader" after you have pressed the button on the Teensy (Stage 1)?

Unfortunately, there are currently no freely available complete hex files. 
I want to prevent third parties from commercially distributing devices with my firmware without permission.

There will probably soon be a closed area where registered users (registered devices) can download the full hex files, among other things.

OvErDrIvEwItHdIsTo, 2024-05-10:
When I try to upload the latest firmware with the S&C Teensy Loader in it (ready to update flashing), it's always the same message:

[image]

But now if I try to install without the teensy loader, I can use the Reset and Flash option but at the end it's the same message again. The same if I press the teensy button...

[image]

[image]

[image]

[image]

[image]

I also try another COM port, another browser, another version of the firmware, with loader, without loader...

I don't understand why I can use calibration function,flip,dac,display, format, ect... but not the Flash/update of the same page...

Thanks again for your time.
Peace

OvErDrIvEwItHdIsTo, 2024-05-10:
I found the old .HEX firmware at this address:
https://github.com/eh2k/squares-and-circles/files/7872600/firmwares_00g.zip

eh2k, 2024-05-10:
Hi @OvErDrIvEwItHdIsTo 

Let me summarise.
 * The flashing of the loader has worked (you see the splash screen with the flashing Ready to upgrade)

In my case it looks like this (with a small browser window, otherwise I can still see the image of the module):

  * In your screenshots I noticed the message ‘Error with Permission-Policy-Header...’ - does this always appear for you?

[image]

If I click on `Flash/Upgrade`, I get the following:

[image]

I have added more tracing to the console output, can you please tell me exactly what appears in the line ‘memMap...’ before you get the "Uncaucht TypeError" ...

To be on the safe side, try disable caching and press ‘CTRL+F5’ a few times.

[image]

Thanks for the hint with the hex file (It is a very old ‘unstable’ version).

OvErDrIvEwItHdIsTo, 2024-05-12:
Hi.
The message ‘Error with Permission-Policy-Header...’ appear only when I refresh the webflasher page or with ‘CTRL+F5’, with the S&C connected.

When I press ‘CTRL+F5’ I've got this message:

[image]

And when I click on Flash/Upgrade, with disable caching, I've got this: 

[image]

[image]

Without refresh or ‘CTRL+F5’ I've got this:

[image]

Excuse me with my English if I don't answer correctly to your solutions.

eh2k, 2024-05-12:
@OvErDrIvEwItHdIsTo 

Thanks again for the reply. I think you are using an outdated version of chrome.

The `toArray` function seems to be available from [122 upwards](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Iterator/toArray#browser_compatibility).

Please paste `chrome://version/` into the address bar to check your Chrome version. (You seem to still be using the 109 version, which is over a year old).
It is generally recommended to update chrome.

OvErDrIvEwItHdIsTo, 2024-05-13:
I can't update Chrome since a long time because of my version of Windows, until today I've got no problems because I'm using teensyloader or Arduino IDE, STM32CubeProgramer, Thonny, ect ect... 

It's my first time with WebFlasher.

Sorry for my incompetence, I will trying to borrow a better computer and see if it's ok. 

Thanks

OvErDrIvEwItHdIsTo, 2024-05-14:
Noob noob noob, I'm an old Noob !!!! 

Everything is alright with a recent computer. 
Now I have to calibrate it. 

Thanks for everything, S&C is my best Eurorack Audio Module.

eh2k, 2024-05-14:
That's good, btw. I also learned something and was able to improve a few things. Otherwise, you can also do updates from your smartphone (testet with chrome - if you have a suitable cable/adapter).

LPC1972, 2024-05-16:
Hi, first of all thanks for your amazing work on S&C, I use it literally in every patch! But unfortunately I have the exact same problem as [OvErDrIvEwItHdIsTo]("toArray"-function error), because of audio interface/FW driver it's complicated to update my OS with the result that I can't update Chrome for the moment. Hence, I can't flash my S&C anymore... Currently i'm on Chrome version 116. Isn't there a work around for now so at least I can get my S&C back up and running?

OvErDrIvEwItHdIsTo, 2024-05-16:
I do not success with my smartphone, even if I got the good version of chrome and the good cable.
The only solution was to use an other computer, so I went to the library with my S&C to use better one. Great Success.

eh2k, 2024-05-16:
@LPC1972 

I have recently removed the [toArray](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Iterator/toArray#browser_compatibility) call. Please refresh the webflaher with CTRL-F5 and try again. As long as its in my code, I can try adjust something. If it doesn't work, please send a screenshot with a detailed error description. As far as testing is concerned, it works on all my devices :-(

LPC1972, 2024-05-17:
> @LPC1972
>
> I have recently removed the [toArray](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Iterator/toArray#browser_compatibility) call. Please refresh the [...]

Yes, that did the trick, that was easy!! :) Thanks again for your great work and super fast help!

mrmattallison, 2024-09-11:
Hi @eh2k 

I went to update my O&C today and similar issues to others, the web flasher isn't working for me.

It finds the teensy (with bootloader, old firmware wiped in the process), but just gets stuck in a loop of being able to only upload the bootloader, not the firmware, with the only button being "Calibration", which works, as does midi and I can send and parse bytes like 0xF3 0x4C to open and close the machine engine.

[image]

[image]

I'm running the latest version of Chrome, with web midi enabled on MacOS 11.7 on ARM64.

Attached are some screenshots. I think there is. bug in the code and have attached the screenshot for the JSON error.

[image]

A friend (with forewarning it might wipe his unit) tried to update on a Windows machine and ended up in the same place I am with a unit that's stuck waiting for an update, that can't update.

Any idea on how I/we can resolve this? Thanks in advance!

guytaylor, 2024-09-11:
Hi @eh2k,
I am the guy Matt (above) mentioned, and it's done the same to mine...

HeWhoWantsJeansAgain, 2024-09-21:
Update from my previous post - my issues might have been _on me_. I was using Cubase to do some recording while doing my update. After a few attempts of getting stuck at "Ready to Update" I thought it was odd that after flashing the Loader, the Webflasher would show "No midi device found" (something to that extent). So I decided to finish my work, exit Cubase and try with nothing running - figuring maybe on the reboot Cubase was grabbing the attention or... something.

Anyhow, it worked _fine_ with nothing running but Chrome. It's possible Cubase got in the way, but I cannot confirm that. But I've also now successfully updated / flashed two o_C units in a row so that seems like a good indicator.

eh2k, 2024-10-27:
With the firmware version e030c30 there seems to be a bug in the update process with the webflasher. I can't say exactly when this started and whether it came in with a Chromium update. On my computer, the midi transmission stops at a random position during the update while the Chrome tab also stops responding.

In this case you should press the reset button on the Teense and install the squares-and-circles-loader (Webflasher Stage 1).
After that the update should run continuously until the end.

Background info: It is somehow related to the `usbMIDI.send_now()` function calls while receiving midi data. If I do not call it, the Midi connection remains stable.

closed 2025-05-20 by eh2k