eh2k/squares-and-circles #68: 707/909 have problems to flash in the latest

koswir, 2024-03-18:
Hey,
After a few times they have flashed, but before that I had some errors:
[image]
[image]
[image]
Once I even got that black screen and o_c stoped to load. 
I had to flash s&c again!

eh2k, 2024-03-18:
Thanks for reporting - if this happens often or in general, reducing the transmit_frame or increasing the transmit_delay will probably help.

To do this, simply adjust the following parameters in the URL of webflasher:

default:
- `https://eh2k.github.io/□●/flash/...../?&transmit_delay=1&transmit_frame=64`

slower but safer:
- `https://eh2k.github.io/□●/flash/...../?&transmit_delay=1&transmit_frame=16`

It may also be that the problem occurs depending on the engines loaded. It's best to set all engines to GND when flashing (this will happen automatically in the next update)

koswir, 2024-03-18:
Thank you. I am going to test it! 💯

closed 2024-10-28 by eh2k