eh2k/squares-and-circles #69: Poly FM synth

koswir, 2024-03-18:
Hey @eh2k 
In the Midi section there is VAx6 synth (I like it alot!)
Any chance for DxFMx6 version?
Pleeeease! 😺

eh2k, 2024-03-18:
That's still on my to-do list (midi control). 

If you haven't already discovered it, I added a stereolize mode/config in the io-config to DxFM some time ago, i.e. two voices are used alternately.

koswir, 2024-03-20:
> That's still on my to-do list (midi control).

Great!

> If you haven't already discovered it, I added a stereolize mode/config in the io-config to DxFM some time ago, i.e. two voices are used alternately.

I've noticed that stereo alternating option, but would be much cooler to have 6 voices on one part!

eh2k, 2026-10-31:
@koswir - finally done: in the 2026-10-31 - 3e930ee release SYNTH/DxFM is polyphonic via MIDI with 8 voices (plus the two Trig/CV voices), with sustain (CC 64) and a soft limiter, so chords don't clip. SYNTH/Plaits got 4-voice MIDI polyphony as well; both replace MIDI/VAx6 (its patches still load).

closed 2026-10-31 by eh2k