eh2k/squares-and-circles #74: Enhancement idea for the Patch storage.

koswir, 2024-04-15:
I love the Patch storage - it is amazeing!

Would be possible to copy settings from one machine to an another?

I use s&c for poly patches and it takes alot of time to configure all the voices!
Anyway even without copy/paste the Patch storage will be super helpful for creating some templates!
Thank you @eh2k!

eh2k, 2024-04-15:
@koswir - thanks.

Forgot to mention: If you run the `FORMAT SD` in WebFlasher, all patches (except the last loaded/saved one) are gone.
And you can go back to the previous version - the current loaded/saved patch remains active.

I'm thinking about adding the possibility to download all saved patches or a general backup mechanism (sysex dump).

----
Regarding poly patches - I had once a "experimental" CopyCat engine (which I had discarded).

The idea was e.g. to select a first engine, and then set 1-3 CopyCat engines. If you changed the first engine, the others were also replaced dynamically (UseCase was basically to configure multiple voices. The paramters of the CopyCat were only offsets relative to the first). But in the end it was all very complicated - and the approach didn't quite work out, e.g. with polyphonic MidiEngines like VAx6.

koswir, 2024-04-15:
> @koswir - thanks.
> 
> Forgot to mention: If you run the `FORMAT SD` in WebFlasher, all patches (except the last loaded/saved one) are gone. And you can go back to the previous version - the current loaded/saved patch remains active.

Good to know. Thanks.

> I'm thinking about adding the possibility to download all saved patches or a general backup mechanism (sysex dump).

That would be great! 

> Regarding poly patches - I had once a "experimental" CopyCat engine (which I had discarded).
> The idea was e.g. to select a first engine, and then set 1-3 CopyCat engines. If you changed the first [...]

Actually I have been thinking about something like that, but it seems complex even to write a feature idea about it. 😄 

What about something like "COPY Machine from x" on the machine-selection-page?
[image]

eh2k, 2024-04-17:
I"m not shure if the following behavior is generally known, so here is a short explanation:

If you select a new engine in the Machines/Engines page by short-presssing-right-encoder then the last IO settings are only kept under certain conditions.

For example, if you change from a fx engine to another fx engine, it would be a pain if the AUX mixer settings were reset each time. On the other hand, if you switch from a fx engine to a synth engine, there are no  parameters to take over (e.g. quantizer) - in this case, a reset to a default occurs.

I think this can be confusing, but it is the same behavior since the early versions.

And there are cases where you simply want to reset the default IO settings after changing engines. That's why I also handling the long-press-right-encoder when selecting an engine - here IO-Config is always initialized to default (this is also quite useful for debugging).

I think this can be improved as follows:
Instead resetting IO settings to default - i.e. a popup with the following choices would appear:
  * `Reset IO-settings` (default at the moment)
  * `Try keep IO-settings` (default short-press)
  * `Copy from X` (All Parameter-Values + Modulations + IO-Config will be copied)
    * Presumably copying the paramters+modulations only makes sense with same/compatible engines ?!
    * When copying the IO settings, a few special cases will probably be necessary or a mixture of `Reset IO-settings` and `Try keep IO-config`.

Perhaps even more operations ...  suggestions welcome.

koswir, 2024-04-18:
> I"m not shure if the following behavior is generally known, so here is a short explanation:
> If you select a new engine in the Machines/Engines page by short-presssing-right-encoder then the last IO settings are only kept under certain conditions.
> And there are cases where you simply want to reset the default IO settings after changing engines. That's why I also handling the long-press-right-encoder when selecting an engine - here IO-Config is always initialized to default (this is also quite useful for debugging).

Wow! I didn't know about it - it's very handy!

> I think this can be improved as follows: Instead resetting IO settings to default - i.e. a popup with the following choices would appear:
 > * `Reset IO-settings` (default at the moment)
> * `Try keep IO-settings` (default short-press)
> * `Copy from X` (All Parameter-Values + Modulations + IO-Config will be copied)
>   * Presumably copying the paramters+modulations only makes sense with [...]

I love the idea!
I've been contemplating multiple copy choices for special cases, but that might be too complex and messy. 
I don't think it's necessary. 
In general, everyone is used to a copy command creating an exact duplicate that can be edited later. 
In my opinion, copying with All Parameter-Values + Modulations + IO-Config (if possible) is perfect!

> ![image](https://private-user-images.githubusercontent.com/11786292/323355798-b4c04b5c-38ab-4c42-bf4d-0d99dbd7af97.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM0NzMwMDgsIm5iZiI6MTcxMzQ3MjcwOCwicGF0aCI6Ii8xMTc4NjI5Mi8zMjMzNTU3OTgtYjRjMDRiNWMtMzhhYi00YzQyLWJmNGQtMGQ5OWRiZDdhZjk3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE4VDIwMzgyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFkYmM2MzAwNjRiNzBiOTE0ODE2NGEwMmM0MGFiNTdhNmFhYmUyNGNkZTA1NmUzN2MyYWUyZmYzYzVjNzE1NTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hujpNbwF7wBi4NrmY4ZUzE8NlW8PZMQS_czFamkTxx0) ![image](https://private-user-images.githubusercontent.com/11786292/323356389-51ea9d09-2489-4f86-89c9-4405e46d35fc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM0NzMwMDgsIm5iZiI6MTcxMzQ3MjcwOCwicGF0aCI6Ii8xMTc4NjI5Mi8zMjMzNTYzODktNTFlYTlkMDktMjQ4OS00Zjg2LTg5YzktNDQwNWU0NmQzNWZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE4VDIwMzgyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmYTg5OWQ5ODhlZWU2MWIxZmNkMmU2ZmNhMTk2MDIyN2JhZDNjNWExMDMzZjhmZTIyYjIwYTg4MjY2ZjcxM2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._X62-HetNSGF13xmT3jD0UXjPpynLYdC1eCwQ0qdhfk) ![image](https://private-user-images.githubusercontent.com/11786292/323355493-170bdb33-2698-4749-9f88-1b3bb553b7e0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM0NzMwMDgsIm5iZiI6MTcxMzQ3MjcwOCwicGF0aCI6Ii8xMTc4NjI5Mi8zMjMzNTU0OTMtMTcwYmRiMzMtMjY5OC00NzQ5LTlmODgtMWIzYmI1NTNiN2UwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE4VDIwMzgyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU5NjU5MTVmY2QzZDRiMGJkY2VmOGQ2OTNhMmIzYjVkMjkwYTgyNjE3NzIyYzk5Zjk5ZTM0Y2VlODQ2NGNiNDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Z257VuRD2GtMTCODIGNSvTMHdvf7yyV3kd3JBV6L9XU)

It looks great!

eh2k, 2024-04-19:
@koswir 

just pushed 36b6694

Two "small" enhancements:
   * SD Format - preserves stored patches
   * Engine Load "Options" popup (reset IO + copy from)
      * I kept it very simple with the copy. The engine is cloned/copied exactly (paramters + modulations + io-settings). 
      * When copying, it does not matter which engine was selected (also working on GND/---)

Thanks for the "Copy" suggestion  - I quite like the way it has now been implemented with the popup.

koswir, 2024-04-19:
@eh2k Thank you. 
I am going to test it over weekend.
I have already noticed plenty of weird behaviour tho - mostly parameters doesn't work or are not copied properly.

eh2k, 2024-04-19:
@koswir 

please make sure that this is version 9b81a8a (I just fixed the copy bug) - and have edited my previous message.

koswir, 2024-04-19:
@eh2k It's all good now!
I have only noticed a small thing - m-osc waveforms shape parameter loads by default with "Shape" name.
The same happens after copy. 

I will test more over weekend!
Take care!

closed 2024-12-01 by eh2k