eh2k/squares-and-circles #75: Firmware won't load

arpeggio1234, 2024-04-23:
A practical question/thing
Earlier i loaded the squares and circles firmware by means of the webflasher.
Now i am trying to load the new firmware and it won't load. I use windows 8.1 and use chrome or opera. Can this be the issue it won't load. Can some one provide a hex file please if this is possible. Thanks for making this great firmware, it sounds great.
Dian from the Netherlands

eh2k, 2024-04-23:
Hi Dian @arpeggio1234 

Please provide more details or any error messages (step by step to reprode)

Is the device recognized as midi device?
You could try flashing the loader.hex first.

_Copied from the we [webfasher](https://eh2k.github.io/%E2%96%A1%E2%97%8F/flash/?firmware)_:
---- 
The flashing process

Stage 1: Flashing Squares&Circles Loader

For a bare Teensy 4.0, the loader firmware must be installed first. This can be done directly in your browser (tested with Chrome) following the steps on the left side.

If your broswer is not supporting the serial interface, you can download the [Squares&Circles loader hex file](https://github.com/eh2k/squares-and-circles/releases/download/0.0.0/squares-and-circles-teensy4XX-loader.hex) and flash it with the Teensy Loader. After flashing the loader the device startups only showing the splash screen with a blinking "Ready to Update".

Stage 2: Flashing Updates + Machines/Engines

If the Squares&Circles device is connected to your PC, i.e. the loader or an older version has already been installed, it is automatically recognized via [WebMidi](https://studiocode.dev/check-webmidi/) and the [Update] button and other functions are automatically activated.

arpeggio1234, 2024-04-24:
Hello, good news.
I installed Chrome on my Linux (Manjaro)
Laptop and it worked fine the firmware update.
Thanks for the support

closed 2024-04-25 by eh2k