eh2k/squares-and-circles #88: Patch storage - corrupts installed TR707/TR909 sample memory

eh2k, 2024-06-06:
> ...at first this wasn’t the case i did have three preset configs saved with no problems , but then i realized that i didn’t have the tr707 & 909 drums in the menu so flashed them onto the teensy , 
 after that I started getting crashes where the screen goes pixelated and sound cuts, also i could save a second preset BUT loading that preset would just bring up the S&C boot screen and nothing was there, 
 thought maybe preset ram/(rom?) was corrupted OR maybe the 707/909 drum samples were taking up space so reflashed the teensy but this time with ONLY the 707 samples...  seemed ok for a while but then another pixel screen glitch and crash ....

Hi matt,

thanks for reporting. I can reproduce the issue and will fix it with the next update.

It looks like the patch saving is overwriting the memory of the TR707/TR909 blobs.

There are two commands in the webflasher:
  * FormatSD (keep patches)
  * FormatSD (complete)

The FormatSD (keep patches) command deletes the memory slots where the optional engines (TR707/TR909) are stored, but keeps the patches. You can flash the TR707/TR909 again after that, so the memory area should be written after the patches and will solve the issue. New patch on empty slots will overwrite the TR707/TR909 memory area again.

hotstuffwonkyair, 2024-06-07:
thanks for the heads up on the SD format thing , so what i’m thinking is best for now is to do an SD format & then save, say, 6 patches all with the 4 slots filled with any of the engines, then do an sd format (keep patches) , then flash the 707 & 909 samples onto S&C , then all should be ok as long as I ONLY overwrite patches into the 6 presaved slots..
and sit tight for a fixed version sometime.
does that sound like a wise routine to go with?
muchos gracias,
yrs, matt

eh2k, 2024-06-15:
@hotstuffwonkyair 

I have just pushed a new version. I have added some extra memory checks and fixed some things that caused crashes in special cases. 

However, I can't really explain the issue you describe - unless the crc checks are reporting memory inconsistencies with the latest version a75d8e.

If you get a crash again, a picture of the crash report would be helpful.

hotstuffwonkyair, 2024-06-15:
great news! :) i’ll give it try tomorrow,  how do i get a crash report , in general the crashes involve the display fragmenting and inability to navigate to anything.. so i generally reboot, is there a crash log kept somewhere on the device i can access? is that via the webeditor interface?  sorry i’m not too familiar with ALL the ins and outs of the platform/code.

eh2k, 2024-06-16:
@hotstuffwonkyair,

could be the same issue - https://github.com/eh2k/squares-and-circles/issues/63
The crash report should be shown automatically on the display immediately after the crash.

hotstuffwonkyair, 2024-06-16:
> @hotstuffwonkyair,
> 
> could be the same issue - #63 The crash report should be shown automatically on the display immediately after the crash.

i did ponder that after noticing it just recently , but my issue seemed to be triggered from saving presets and changing machines never caused a problem for me .
ANY ROAD. have flashed this new version ( full SD format & including the 707 & 909 samples )  and don’t seem to be having the same issue anymore :) ... haven!t really pushed it too far yet ..ie just 2 presets up to now.. i’ll get back after i’ve had a more in depth session with it ..
 muchos gracias ..

themangoest, 2024-07-22:
When it says SD does it mean that it can be saved on the sd card for a teensy 4.1 for example also?

eh2k, 2024-07-22:
@themangoest 

“SD” is currently a memory area in the firmware flash memory and there is no difference with the T40 or T41.

I've called it “SD” as I was thinking of mapping it to an SD card at some point (if the hardware supports it). T40 would therefore have a “virtual SD”.

But as I said, the hardware SD is not yet implemented.

wtkdwc, 2024-09-15:
hi, i think this issue still persists, yesterday i installed new fw with webflasher, just with flip 180 option and 909. saved two patches and when trying to load any of them, the module freezes and restarts

eh2k, 2024-09-15:
hi @wtkdwc   

Can you load the patch after the device has restarted?

Was a crash report displayed after the restart?

wtkdwc, 2024-09-15:
No, it restarts to the second patch where it was before. No crash report, just restart.. see here https://we.tl/t-AFFxLydEaS

eh2k, 2024-09-15:
@wtkdwc - thanks for reporting.

According to the video, you have an old firmware 9b81a8a_fd9f157 on your device. The current firmware has the version: e030c30_a751cf7.

wtkdwc, 2024-09-15:
hmm must have missed sth when reflashing, however i did now also format sd (keep patches) on version e030c30 and behaves the same, will try complete format and creating new patch to test

wtkdwc, 2024-09-15:
seems 909 still clashes even with the version e030c30, after complete format i was able to save and load patches, then i tried to load 909 and load previously saved patch and module restarted same as in my previous video