eh2k/squares-and-circles #89: Missing scale and offset on the LFO

eh2k, 2024-06-10:
>              I'm missing scale and offset on the LFO's. Maybe I just can't figure it out ?

Also - Can you make a LFO like on a moog one (see the pic) ? 
I can't figure out the algorithm - but I use my own wavetable in VCV: https://github.com/jpnielsen/wavetables 

[image]
<hr>
Just as a FYI, I totally gave up on flashing in  windows - the bootloader is flashed, but the "squares&circles" is not recognized - just a timeout in the log. 

[image]

So I finally tried on a raspberry pi5: 
the bootloader did not flash, it immediatley returned with a "flashing complete" message - but didn't do anything. maybe I'm missing some config for the [udev rules](https://www.pjrc.com/teensy/00-teensy.rules) - I will try that later.

When i flashed the bootloader using the teensyloader, the webflasher in chromium recognized the "squares&circles", and got the "update" and it worked, so I'm now capable of trying your latest creations.

_Originally posted by @jpnielsen in https://github.com/eh2k/squares-and-circles/issues/1#issuecomment-2135227560_

mrmattallison, 2024-10-18:
Just wanted to add my voice to the mix. Scale and Offset would be great, both in the Modulation parameter option as we as the CV/LFO machiene. I'd love to see a SMOOTH Random option as well ala MI Peaks.

eh2k, 2024-10-27:
@jpnielsen @mrmattallison 

CV/LFO Enhancements in the latest version

[image]

closed 2024-12-01 by eh2k