eh2k/squares-and-circles #91: Arpeggio or sequencer engine

themangoest, 2024-07-22:
Hello,
Is it possible to have a sequencer and/or arpeggio engine for squares and circles?

eh2k, 2024-07-22:
I have already experimented in this direction, but it is not yet in the firmware - see also  https://github.com/eh2k/squares-and-circles/issues/72.

Arpeggio could also be implemented as a modulation.
However, this would have to have a max of 3 parameters - concrete suggestions are welcome.

themangoest, 2024-07-22:
Sure, if possible parameter for:
- root note
- direction up, up/down, down and random
- range spread (notes stays in one octave or more)

themangoest, 2024-07-22:
So we can use it via ? Or can this be implemented ?

eh2k, 2024-07-22:
It is not yet implemented - but it is basically exactly the proposal

themangoest, 2024-07-22:
Yes, I hope you can do it.

eh2k, 2024-07-22:
I'm currently taking a summer break - otherwise this would be low-hanging fruit.

themangoest, 2024-07-22:
Enjoy you summer break!

themangoest, 2024-09-09:
Do you think it is possible to add the arpeggio as an app?
Thank you

eh2k, 2024-09-20:
Was working on the Arp modulation today...

These are the current parameters: 
 - Trig: !, 1...8
 - Reset: CLK, 1...8
 - Mode: Up, Down, UpDown, Rand
 - Chord: see plaits chord_bank.cc
 - Octave: 1-10
 
I'm thinking about whether the modulation only makes sense for the V_OCT parameter (piano roll) or if it also interesting to assigned on all parameters ?!

Combined with the `CV/Voltage` app, it is easy to control external modules.

themangoest, 2024-09-21:
I would already be happy with only v/oct assignment but redirecting to other parameters is also cool if it is not too much work.

Good job 🤘

eh2k, 2024-10-27:
@themangoest @mrmattallison and other

just pushed a new update...

In the end, I decided against ARP modulation in favour of a generic SEQ modulation:

[image]

There are two new SEQ engines:

[image]

[image]

Regarding the trigger/clock:
If you press both buttons, a menu appears where you can access the CLK/Midi settings - there you can configure a clock signal (4ppqn) on T1.

Technically, this can also be a midi signal. However, the timing is a complex thing, especially if you mix reset triggers with midi clock - I still have a few things on my to-do list (maybe just remove the midi clock support).

The sequencer stuff opens up a lot of new possibilities -  - probably some I hadn't thought of.
As always, feedback and suggestions welcome.

themangoest, 2024-10-27:
Woaw, I'll try it out and will keep you posted

rootapprox, 2024-10-27:
@eh2k where can i find this new update ?

rootapprox, 2024-10-27:
Hello @eh2k  i just updated the new firmware and in the Drums i dont find some of them like FM drum ,  808 .. also i see in the list of engines of the readme other engines not available for me like Linm, DMX , am i missing something maybe? 

Great work for this!!

eh2k, 2024-10-27:
@rootapprox 

thanks for the hint and your feedback. I have just pushed the missing engines: FM-Drum, 808ish-BD, 808ish-SD, 808ish-HiHat (please update the firmware again to fix it). I lost them during the sync with my dev-repo. The Linn,DMX engines are not released yet (they were listed by mistake). The TR707-Hihats does no longer exists. The TR-707 Hihats can be mapped with the TR707 Multi-Trigs (if you set the default trigger to -, you can configure a dedicated trigger for each sample - see https://github.com/eh2k/squares-and-circles/issues/29).

rootapprox, 2024-10-27:
@eh2k 
thanks for the quick update!! 
just another thing im encountering the Modulation LFO type always spawn back to SINE when restarting/OFF-on the module and its not saving the actual LFO type i saved the Patch in Patch storage.

rootapprox, 2024-11-16:
@eh2k Can i ask you please where to find a manual for the Turing Machine SEQ , i dont understand the Out 1 Out 2 with its mutilple modes.  Thanks!

eh2k, 2024-11-18:
@rootapprox - 

I have tried to describe the TM functionality with chat-gpt support. I hope this is understandable. I think the additional routing in the IO page might be confusing here . You can mix both CVs to one CV-Out (I don't know if that makes sense).  Make shure you set the Output to A+B or C+D, so the virtual CV signals are on separate phyiscal outputs.

https://github.com/eh2k/squares-and-circles/wiki/TM

rootapprox, 2024-11-18:
Thankls alot @eh2k for the description of the TM engine , clear manual use. about mixing CVs yes it makes sense, its just if its possible to have the possiblity to use the SEQ engines and assign em internally without using patch cables , ( for saving space to use stereo for the next channel ..)

closed 2024-12-01 by eh2k