eh2k/squares-and-circles #92: had some issues with random distorted scratchy noises coming from the o_c with squares circles.

xanopticon, 2024-08-24:
Hi

had some issues with random distorted scratchy noises coming from the o_c with squares circles. 
I tried updating flashing the firmware (never done before) trough some web application 
https://eh2k.github.io/%E2%96%A1%E2%97%8F/flash/?target=OC_T40&deviceId=04E9E5165D13&transmit_delay=1&transmit_frame=64

But in the end it only resulted in the screen being positioned too much to the right and no sound output at all.

Anyone else who had this issue before? 

thanks!

eh2k, 2024-08-25:
Hi 
regarding display see: https://github.com/eh2k/squares-and-circles/issues/90

what is your exact setup (used engines) where you get "random distorted scratchy noises"? 

Is the calibration is ok? (You can test it with the CV/voltage). With default hardware -3..6V, clipping can theoretically occur at high volumes.

xanopticon, 2024-08-25:
Not sure to be honest. I bought it like this and the builder said it was a software issue so I tried to re-flash firmware on it. but in the end it made no sound anymore at all.
maybe I should factory reset the teensy or something and then flash?
Maybe if i manage to successfully install the firmware the problem might be solved with the scratchy noises and pops.

xanopticon, 2024-08-26:
> Hi regarding display see: #90
> 
> what is your exact setup (used engines) where you get "random distorted scratchy noises"?
> 
> Is the calibration is ok? (You can test it with the CV/voltage). With default hardware -3..6V, clipping can theoretically occur at high volumes.

pretty sure it's broken..
it makes sound but extremely low volume and with a very high noise floor..
volume is set to max and it's not making much noise.. 
also feel like anything I try just makes the thing worst.

eh2k, 2024-08-26:
One option would be if noise occurs despite successful callibration, you could test with a different power supply unit - or disconnect other digital modules from the power temporary to be on the safe side.

xanopticon, 2024-09-02:
> One option would be if noise occurs despite successful callibration, you could test with a different power supply unit - or disconnect other digital modules from the power temporary to be on the safe side.

Tried a complete calibration and also tried another power supply or even it's own case.
problem still remains, even if I turn the channel volume to 0. 
So maybe a faulty device is the only logical explanation?
Have you ever experienced this issue anywhere else?

eh2k, 2024-09-02:
The problem is new to me. If the GND engine is on all four slots, the DAC should have a constant voltage (0V when calibrated). Nothing should be audible.