eh2k/squares-and-circles #96: Bootloader Flash Loop

scuzzphut, 2024-10-12:
Hi - I'm having a problem flashing a Teensy 4.0 via the web flasher.

I can flash the bootloader fine using the process under "Stage 1: Flashing Squares&Circles Loader".   

When complete , the screen on my O_C flashes "Ready to update" but the web flasher returns to the bootloader page (Stage 1: Flashing Squares&Circles Loader)   instead of proceeding to the setup page. 

I've tried this with Chrome and Brave browsers, but no joy.  Any thoughts on what to try next ? 

Thank you !

eh2k, 2024-10-12:
Hi - the second step only works if there is a web-midi connection. 

Please check this, maybe it's the same reason: https://github.com/eh2k/squares-and-circles/issues/66#issuecomment-2342582007

scuzzphut, 2024-10-12:
thank you eh2k. 

I just connected the O_C to my work laptop instead of my desktop (which has multiple midi devices, drivers, utils etc installed)  and the firware flash page appeared. Everything worked ok from there so i now have an operational unit again. 

Another one to add to the best guess at a midi clash of some sorts.