eh2k/squares-and-circles #97: Patch storage recall crashes/reboots module

mrmattallison, 2024-10-18:
Hi @eh2k,

I'm experiencing an issue whereby if I recall saved patches the module reboots reverting back to the previous module settings, not the recalled patch. Any ideas on what could be causing this? It doesn't seem to follow a particular workflow in that sometimes the same patch will load fine, other times it will reboot the module and I'll need to try load another patch, then reload the previous patch.

mrmattallison, 2024-10-18:
https://github.com/user-attachments/assets/be1f1a65-9cd5-4597-9819-c0abc697b20c

Video attached of the stated behaviour

mrmattallison, 2024-10-18:
On further investigation it seems I can recall Patch 1, 2 & 8 without any issues, every time, but the rest are all a gamble.

eh2k, 2024-10-18:
@mrmattallison, thanks for reporting.

Looks like the same issue: https://github.com/eh2k/squares-and-circles/issues/88#issuecomment-2351454620

I'm thinking about adding the possibility to dump the patches as sysex in the next version. I could then analyze them - maybe there is an infinite loop which leads to a reset by the watchdog.

mrmattallison, 2024-10-18:
Ah! Thanks, yeah, seems to be a similar/same issue. I haven’t tried the TR style drums machines then/there, but the save and recall seems to be similar. 

Let me know if there is anything I can test my end to further assist, otherwise I look forward to the next version with the suggested sysex changes.

eh2k, 2024-10-19:
Unfortunately, I cannot reproduce this. I would actually need an exact step-by-step description of how the patch is created. It could also be that there is not enough RAM when loading.

If you press the right button at startup, no patch is loaded. So you could try to load the patch from the scratch. Alternatively, you could try to rebuild the patch, then save the patch and load it again, which might lead to a crash.

eh2k, 2024-11-14:
Probably fixed in 3a6a2c8

eh2k, 2024-11-15:
>I just have a question is it normal that the module reboot when i load a Patch ? from patch storage ?

_Originally posted by @rootapprox in https://github.com/eh2k/squares-and-circles/issues/100#issuecomment-2479040653_

No - it's not normal. 
I have to test this again, I optimized something in the last version.

rootapprox, 2024-11-15:
Thanks a lot for verifying that ! @eh2k

eh2k, 2024-11-24:
@rootapprox 

I think I have found the cause (was something different), or I will fix it with the next update.

Patches cannot be loaded at the current version if they exceed 256 bytes. The size of the patch is dynamic - depending on which engines and how many modulations are configured.

rootapprox, 2024-11-25:
@eh2k Makes complete sense , cause i do use lot of modulations and maybe some designed fxs shared from the other friend here in the group

eh2k, 2024-11-30:
@rootapprox - Hope the loading issue is fixed with 5c87eb5.