eh2k/squares-and-circles #100: Missing reset in the turing modulation 🥺

eh2k, 2024-10-29:
> ...so please add a reset to the turing modulation 🥺 

@koswir - https://github.com/eh2k/squares-and-circles/issues/99#issue-2619831547

koswir, 2024-10-29:
ah right - I didnt know that!
I am used to the VCV version where is the reset!
It resets shift register's position to it's step 1 so the locked sequence can be easly synced with other sequencers.
[image]

eh2k, 2024-10-30:
@koswir,

I have tried to understand the reset in the vcv rack - and to be honest, I have no idea what's going on with the reset trigger...

Technically speaking, the Turing machine has two states. One is the 16-bit shift register and the other is the seed for the random function.

Currently at startup, the shift register is initialized with 0 and an random seed (system ticks) is set for each turing machine, i.e. the random sequence is different for each turing machine.

Since with each clock the last bit is randomly flipped and set as the first bit again, you cannot actually play back the sequence (without remembering all the random numbers and the prob values etc).

Logically, you can perform only a reset by setting the shift register to 0 combined with a fixed seed value. Consequently, the random sequence would repeat exactly the same.

Unfortunately, I cannot introduce a new parameter for the seed, so I would hardcode a const value. As the propability param only goes from 0..9 - you would then have a maximum of 10 different start sequences which are infinitely long and change depending on the step parameter...

To summarize, there are cases where a reset is useful so I would implement this as described. I.e. if you leave the reset at ‘-’ you would have the old behavior... (like the real tm)

koswir, 2024-10-31:
I mostly use tm to create a locked loop I like - and than I keep it repeating unless I need to change a pattern.
I am not sure do I understand it correctly but wouldn't the reset you described ruin the locked loop with a new one?

koswir, 2024-10-31:
Ok I have the idea.
Normally, the Turing Machine shifts the bits, introducing randomness at the first bit, which influences the entire sequence. 
When the Turing Machine "locks" its state, it stops randomizing the first bit of the sequence. Instead, it cycles the existing sequence repeatedly. But when locked, it copies the last bit to the first bit, maintaining the same sequence in a loop. This way, the locked state prevents new random values from entering the sequence, cycling the current state continuously.

So could you store the state of the 8 bits when the sequence starts to be locked? (a user switches PROB to zero)

This way when the reset is activated, the sequence would restart from those saved eight bits, returning to the start of locked state. This would give a consistent starting point each time when triggering a reset.

eh2k, 2024-10-31:
Hmm, I hadn't thought of the prob = 0 case. I have to try/experiment saving the shift register and the seed in general when a parameter is changed and use it for reset or patch recovery...

eh2k, 2024-11-02:
@koswir - added the reset with 39a9940. (Also some EuclidArp enhancements)

The TM state is saved on a PROB or STEP param change (and is stored in the patch). You can set a reset trigger to restore the state.

koswir, 2024-11-02:
Thank you!
It works exactly as I wanted!
I love it 🤟💯
Now I can save locked sequences and they repeat and reset prefectly!

eh2k, 2024-11-06:
@koswir,

Another experimental approaches with the turing machine. I wanted to test what it would be like as a dedicated engine.

In contrast to CV/voltage + tm modulation, you could have a trigs out, similar to the trigger expander (Pulse param "1+2" would set the trig out if bit 1 or 2 are true). 
Clock and reset would work like with the other SEQ engines (saving the shiftreg & seed, like discussed here). 
CV out would be 0-5V (+Transpose in the io-settings). One param could be the quantizer (or maybe mapped to chords).
I'm wondering whether I should discard this approach or whether it might be useful in some way.

[image]

koswir, 2024-11-08:
@eh2k so cool!
> @koswir,
> Another experimental approaches with the turing machine. I wanted to test what it would be like as a dedicated engine.

Would be amazeing to have TM as a dedicated engine. Even for controlling internal machines I can see some uses. 
For a dedicated engine its possible to modulate parameters - eg. use external cv for probability or to change a quantizer's scale.
(actually I've been thinking it's a bummer that all the parameters in i/o are not modulable... but that is a different topic)

> In contrast to CV/voltage + tm modulation, you could have a trigs out, similar to the trigger expander (Pulse param "1+2" would set the trig out if bit 1 or 2 are true). Clock and reset would work [...]

Trigger out would be very cool (especially with logic selection from the expander).
Maybe both outputs could have output type selection?
Like out 1: CV, CV quantized, CV inverted?, trig 1, trig 1+2, trig 3 etc.
[image]

I would prefere a quantizer over those preset chords, 😄 
but in general I wouldn't use built in quantizer.

eh2k, 2024-11-14:
SEQ/TuringMachine introduced with 3a6a2c8

rootapprox, 2024-11-14:
Excellent job! can i ask to have a Quantizer ( Braids Quantizer that is in the engines already ) as individual engine ?

koswir, 2024-11-15:
@rootapprox
I've noticed you asked about a quantiser as a separate engine somewhere but I can not find this comment 😄 
The CV engine can do exactly that - just use external pitch cv and set a scale in the i/o settings!

koswir, 2024-11-17:
Hey @eh2k 

I’ve been testing the Turing Machine over the weekend, and it’s great – I haven’t encountered any bugs. I especially love the automatic clock reset when using MIDI or when the clock is set to T1 (also in the Euclid machines). It would be amazing to have this feature in the earlier version of the TM used as a modulator, when the Trig input is set to 'CLK,' allowing it to reset when it stops receiving a MIDI clock or T1 signal.

I also love that you’ve added all the gate variations, including OR and AND for bits, plus the CV inverter. Super cool and works flawlessly!

I’ve been thinking about the built-in quantizer and CV output range. Right now, the output of the TM engine generates a wide range of values, and there’s no internal attenuation before the quantizer. This means that if a user wants to limit the range of notes (let’s say to a 2-octave range), the only option is to disable the built-in quantizer and use an external attenuator, because attenuating a quantized signal would ruin a selected scale.
Would it be possible to add some optional CV ranges for Out1/Out2? For example:
CV0–1 (1v range), CV0–2 (2v range), CV0–3...
I’ve seen something similar in the Hemispheres engine, Enigma Junior, where it offers selectable note-out bit depth:
Note-3: 3-bit depth, 8 notes of the selected scale
Note-4: 4-bit depth, 16 notes of the selected scale
(...)
Note-7: 7-bit depth, 128 notes of the selected scale
https://firmware.phazerville.com/Enigma-Jr

Lastly, for your new built-in sequencers, internal routing of gates and CV signals between engines - similar to audio routing - would be a fantastic addition. Imagine routing gates from different TM bits to the 707 engine!

Thanks so much for your fantastic work!

rootapprox, 2024-11-17:
@koswir can you explain please the TM engine ? whats the OUT 1 and OUT 2 and whats the Inputs outputs of the engine.. i searched there is nothing that can be a manual for it . cheers!

koswir, 2024-11-18:
@rootapprox 
basically it is an implementation of the Music Thing Turing Machine
https://www.musicthing.co.uk/Turing-Machine/
Out1 and Out2 allow you to select the type of signal to send from the engine’s outputs (configurable in the I/O page), such as CV or all the gate outputs from the Pulses expander.
https://www.musicthing.co.uk/Turing-Pulse-Expander/

rootapprox, 2024-11-18:
thanks for replyung! In this engine ? there is 2 outputs ? i open the enginer in channel 1 i have clock as t1 then c1 is what ? and output ? i know whats a turing machine its just i couldn't figure out the menu section in the engine...

koswir, 2024-11-18:
[image]
T1-4 are trig inputs
C1-4 are cv inputs
yes engines can use one or two outputs - go to the bottom of i/o page and select outputs (letters A, B, C, D)
for example A+B will work as two outputs and will send what you set in Out1 / Out2 parameters

closed 2024-12-01 by eh2k