eh2k/squares-and-circles #99: 2024-10-28 version's feedback

koswir, 2024-10-29:
@eh2k 
The new version 🤯🤯🤯
so far I found a scope bug - it doesn't load
[image]

EuclidRythm kicks ass especially with all the modulations! and it got reset input!
The new mod seq is lovely and it got the reset too!
...so please add a reset to the turing modulation 🥺

testing in progress!

koswir, 2024-10-29:
The Arpeggiator chords seem the same as in the Chords engine.
The first problem - all the letters are uppercase - this way minor and major chords look the same. Maybe you can change that to MI MA or MIN MAJ.
The second problem - I think that the output of the arp is kinda out of tune - basic minor and major triads do not sound right. The first note of a chord seems fine but next ones seem out of tune (more like v/oct scale problem than offset).
I can check the calibration of my o_c's but Im pretty sure they are well calibrated (and also the first note seems right).

eh2k, 2024-10-29:
@koswir - thanks for testing.

* The scope will be fixed in the next update (today).
* The chords names is a bit unfortunate. The small font is not case-sensitive, and there is no place for the big font. 
* I have to check the tuning...should be this line of code: https://github.com/eh2k/squares-and-circles/blob/main/app/SEQ/EuclidArp.cpp#L303

I'm not sure whether the EuclidArp is really finished. Somehow I don't quite like it. For example, the SLIDE parameter is currently a simple on/off switch .. alternatively as ‘modulo’ is not intuitive. Without the SLIDE is not good either.

koswir, 2024-10-29:
> @koswir - thanks for testing.
> 
> * The scope will be fixed in the next update (today).
> * The chords names is a bit unfortunate. The small font is not case-sensitive, and there is no place for the big font.

so maybe you could squeeze MI MA there

> * I have to check the tuning...should be this line of code: https://github.com/eh2k/squares-and-circles/blob/main/app/SEQ/EuclidArp.cpp#L303
 
I would love to help but I'm no coder 😞 

> I'm not sure whether the EuclidArp is really finished. Somehow I don't quite like it. For example, the SLIDE parameter is currently a simple on/off switch .. alternatively as ‘modulo’ is not intuitive. Without the SLIDE is not good either.

actually I love the slide - I modulated it with the TM and it was dope to randomize slides

koswir, 2024-10-29:
BTW the pitch issue reminded me that I was about to ask, 
could you please add fine tuning for generators in the i/o? 🥺 
Just below TRANSPOSE
Sometimes it's very handy to tune or detune oscs 🎹