eh2k/squares-and-circles #103: Crash reports on latest Firmware

mrmattallison, 2024-11-02:
Hi @eh2k,

Off the back of some of the fixes in the latest Firmware I updated after having to flash the bootloader again and am now getting constant crashes.

Screenshot of the message attached, also reboots occasionally and displays “Out of Memory” even without a patch loaded.

[image]

Hope it’s helpful!

eh2k, 2024-11-03:
Thanks for reporting.

Unfortunately I can't reproduce this. You have the latest firmware, the checksum is correct (the same on my device).

On a crash report, you can press the right button to proceed without restoring the patch. Are you able to reproduce the crash after that? (You can also press the right button on startup to skip loading the patch). An exact step-by-step instruction would help (which engines are used?).
Are you using USB-Midi?

mrmattallison, 2024-11-03:
Hi @eh2k 

Yes, I should have noted that I did the skip loading saved state option, I even deleted the saved presets via the web tool incase there was an issue there.

The crashes persisted. Crashing with that screen report every-time I tried to change Machiene, it would reload and flash an “Out of Memory” screen on boot, and the same issue, everytime I tried to change the Machiene it would freeze, I’d get that crash report and then reload. Endless cycle.

I reverted back to the previous version of the Firmware and the issues resolved themselves. No crashes, no reports and or out of memory issues.

Not sure what would have changed? Incidentally I didn’t need to reflash the bootloader either, just pushed the update (unlike the latest that would stop the upload at various parts of the upload).

mrmattallison, 2024-11-03:
I meant to add, no, I’m not using USB-midi on my device.

eh2k, 2024-11-03:
@mrmattallison 

The version before my last commit is now listed in the Webflasher. Can you please tell me if you are experiencing the issue with the version ([main_432cb0f](https://eh2k.github.io/%E2%96%A1%E2%97%8F/flash/?select=main_432cb0f))?

With regard to out of memory message - there are different ones. Maybe a short video will help me.

If you have reverted to e030c30 - you will probable have to flash the laoder first (see: https://github.com/eh2k/squares-and-circles/issues/66#issuecomment-2440087144 )

mrmattallison, 2024-11-03:
Hi @eh2k 

I've just reflashed the Teensy with main_432cb0F, the issue of out of ram happened straight away on reboot. It briefly flashed  <<<< Out of RAM >>>>. It loads up with the  808ish-BD in Machine slot 1, there are no other machines loaded in Slots 2-4.

If I hold down the button to change the machine on slot one, I get the crash report (images to follow), a click of the button again displays the <<<< Out of RAM >>>> message again (see image). 

I cannot add any other machines in slots 2-4 without it causing the same crash and <<<< Out of RAM >>>> message.

I held down the right button on reboot to skip the saved patch recall, this worked. 

I added a "Wavesforms" engine in Slot 1, saved a preset, then rebooted. That worked, no crash, no <<<< Out of RAM >>>> message. I added 2 other synth machines to Slots 2 and 3, still good. Saved, rebooted. All good.

I then added the 808ish-BD in Machine slot 4. It loaded, but the moment I try change it, or save the patch, it crashes again with the same crash report.

Skipping the recall saved patch again, then loading up the previous saved patch with the 3 synth machines worked fine. No crashed or <<<< Out of RAM >>>> reports.

I then added different drums engines to machine 4, saving, then rebooting the module. No issues. 

I then loaded the 808ish-BD back into Machine slot 4 and tried to save, and received a crash report again.

It seems to be localised to that 808ish-BD machine.

Hope that helps!

mrmattallison, 2024-11-03:
[image]
[image]

eh2k, 2024-11-06:
@mrmattallison - short update on this.

have now investigated this in more detail and tested it more intensively. The crash could also be caused with the CV/EnvGen_ADSR and FX/JU60_Chorus. 

I have refactored the sources a bit, the whole thing is a bit tricky (heap corruption?!). In the end it could also be a compiler bug. In any case, the gcc parameter [--no-inline](https://github.com/eh2k/squares-and-circles/blob/main/app/DRUMS/808ish-BD.cpp#L26) seems to solve the issue reliably.

In any case, I could no longer reproduce any crashes in the last version - dc35138.

closed 2024-12-01 by eh2k