eh2k/squares-and-circles #104: Weird noisy clock/encoder click bleed

koswir, 2024-11-03:
I've encountered something strange and reproducible on all four of my O_C modules, though not in every ABCD machines combinations. When the scope is set to slot A, it mostly works fine. However, the issue commonly happens when the scope is on slot B and an input, either external or internal, is routed to the scope from slot C.

In the video, there’s a noticeable background noise, which is amplified as I pushed the levels to their maximum to make the issue clearly audible. The sound is recorded via line-in.
https://github.com/user-attachments/assets/4268f939-4909-4c75-bfbd-fab58377c298

eh2k, 2024-11-03:
@koswir - thanks for the hint.

Can you tell me if the AUX input on the Scope is set to one of the CV inputs (and the level is up) in your case.

If so, I can also reproduce this. If I set the AUX input to ‘-’, no "digital" noise can be heard.

Unfortunately, the internal ADCs are very sensitive to display changes, which is also audible. 
This is of course a worse case for the scope (also FFT). If you only want only to see the signal, you can set the scope output to ‘-’.

koswir, 2024-11-04:
@eh2k 
I've spent some time testing, and I think I've narrowed down the issue.

Initially, I assumed it was related to the specific A, B, C, or D channels, but that turned out not to be the case. Here’s what I’ve observed that should help reproduce the issue more consistently:

1. The problem seems to occur when one of the CV inputs (C1-C4) is set as an audio input for an effect (like one of fv1 fxs) , even if it’s not routed directly to the oscilloscope. However, the output of this effect should be connected to speakers (on this channel clicking will be the audible).
2. There needs to be an oscilloscope active on one of the A, B, C, or D channels, and it should be displayed on the screen.
3. Additionally, some kind of audio signal should be going to the oscilloscope. This doesn’t necessarily need to be an external input or an effect routed directly to it - it can be an internal oscillator or voice routed internally from any of the A, B, C, or D channels. (This might be why I initially thought it could be an internal issue - I had probably an fv1 effect patched to my mixer from other channel at the time).
When these conditions are met, a clicking noise becomes audible on the channel where the fv1 effect is active.

eh2k, 2026-10-31:
@koswir - the 2026-10-31 - 3e930ee reworks the CV input path (less ADC load, averaged calibration, hysteresis).
Feel free to reopen if it doesn't work for you.

closed 2026-10-31 by eh2k