eh2k/squares-and-circles #107: Weird crash

rootapprox, 2024-11-16:
[image]

rootapprox, 2024-11-16:
also i cant load any of the Patches i have in the patch Storage, always reboots ...

rootapprox, 2024-11-23:
also unable to restore saved patches .. always takes me to the actual patch..

eh2k, 2024-11-24:
@rootapprox 

i think there are two issues here, one is this https://github.com/eh2k/squares-and-circles/issues/97 (patch restore - reboot without crashreport).

The other should be the crash (as shot in your crashreport).
In the crash report I can see exactly which engines you used. Unfortunately I have not been able to reproduce it - according to the code-address it should have crashed in the IO-page.

Can you actually reproduce the crash reliably?

rootapprox, 2024-11-25:
Yes , i can make the crach everytime i want to load a saved patch , it always takes me to the actual patch,

closed 2025-10-25 by eh2k