eh2k/squares-and-circles #110: M-OSC / Waveforms

rootapprox, 2024-11-21:
Hello there ! is it possible to have the Waveforms Engine to be updated to the this alternative cool firmware code from Braids Renaissance ( https://burns.ca/eurorack.html ) 
Cheers!

eh2k, 2024-11-22:
Hi @rootapprox 

There is the SPEECH/SAM engine. It's basically the same that was added in the Renaissance fork. There are other minor adjustments/enhancements that are not entirely clear to me. Can you tell exactly what you mean - I could look at it in detail then.

rootapprox, 2024-11-22:
Hey @eh2k thanks a lot for the reply! i would love to have the new Chords Mode (The original WTx4 algorithm has been replaced by a more featureful WTCH algorithm)
Chord Modes
Braids Renaissance adds a total of 10 new chord algorithms.

\ \CH, -_CH, /\CH, SICH, WTCH
With the quantizer disabled, these modes play the same chords as the original paraphonic WTx4 algorithm, where the Color parameter blends between 16 different chords. But when quantizer is enabled with one of the diatonic modes (dorian, aeolian, etc), the chords will stay in key, picking the correct major or minor chord and extensions based on the configured quantizer mode and root note. For example if the quantizer is set to C Dorian, sending a D value into the V/oct input will result in a minor triad. Color parameter still controls the chord extension but is also picked correctly based on the scale determined by the quantizer.

\ \x6, -_x6, /\x6, SIx6, WTx6
6 oscillators starting at the 1v/oct input, spaced evenly across the currently selected quantize scale. Color controls the number of scale steps between oscillators. When the Braids quantizer is turned off, the oscillators are evenly spaced by semitones (controlled by Color).

Timbre parameter
The Timbre parameter morphs the sound differently depending on the chord waveform.

For saw algorithms (\ \CH, \ \x6), each note is actually 2 saw waves, and Timbre controls the detuning between the 2 saw waves.

For square algorithms (-_CH, -_x6), Timbre controls the pulse width of the square wave.

For triangle and sine algorithms (/\CH, /\x6, SICH, SIx6), Timbre controls the amount of wavefolding to apply to each oscillator before summing.

For wavetable algorithms (WTCH, WTx6), Timbre morphs between a small set of wavetable entries. This is the same wavetable list as the original WTx4 mode.

eh2k, 2024-11-30:
@rootapprox

New "Braids Renaissance Chords" in M-OSC/Waveforms with 5c87eb5

I don't know if you discovered the ‘stereo’ parameter in the IO-page (available on some engines). If you turn it up, two separate braids instances are used for the left & right channel. In stereo the chords sound even fatter.

https://github.com/eh2k/squares-and-circles/tree/main/lib/braids#braids-renaissance-chords-httpsburnscaeurorackhtml

rootapprox, 2024-11-30:
thanks a lot , Works great ! lovey new chords modes ! with the new update i didnt have to re upload the 707 , 808 and the Fxs stored before ,the dx7 file.. which is cool 
☐DAC_8564 / ☑DAC_8565 can i ask you please about whats the difference here ?

eh2k, 2024-11-30:
@rootapprox - there are devices that use a different DAC chip. The chips are pin-compatible, but must be addressed differently in the software.(https://github.com/eh2k/squares-and-circles/issues/13)

rootapprox, 2024-11-30:
@eh2k  Do you know how i could possibly know which DAC i have ? 
i have this Oc : https://www.tlm-audio.com/en/product-page/oc-t4-0-tlm-audio-eurorack-module-ornament-crime-teensy-4-0

eh2k, 2024-12-01:
@rootapprox if the output is working and you see ☐DAC_8564 / ☑DAC_8565 - it's the 8565. On backside of your OC - there is a chip in the center of the PCB there you can see DAC 8565 printed.

closed 2024-12-01 by eh2k