eh2k/squares-and-circles #113: Multi Trigger Drums

rootapprox, 2024-12-01:
Hello ! great work so far fo this amazing firmware ! can i ask please to have the same TR707 drum engine with TR909 and TR808 kit ?
Cheers!

eh2k, 2024-12-02:
@rootapprox - 

I have a few things on my to-do list that go in this direction. With the existing 909 and 808 drums this doesn't work as well as with the 707 samples because the instruments/models have different parameters.

I was thinking maybe something like this - https://rc-808.com/.
You could load your own patches similar to FV1 or DX7.

rootapprox, 2024-12-02:
@eh2k sounds good ! looking forward to have more multi trigger drums kits

rootapprox, 2025-02-21:
Hello @eh2k  Hope everything is great with you ! is there any update regarding multi shots drums ?

eh2k, 2025-02-24:
@rootapprox 

thanks - unfortunately I don't have much time at the moment - but I will definitely address this next.

As a preview, just have a look at the rc-808 (http://rc808.xsrv.jp/wp/download/). The idea is that you can use it to create drum kits that you can load directly on your S&C device.

If you want to prepare and share/contribute rc-808 drum-kits, just publish them on github and post the link here.
Both file formats PLG (xml files) or standalone (allprm, prm files) are supported.

eh2k, 2025-02-24:
Loading will be In the same way as with https://eh2k.github.io/□●/fx.html

rootapprox, 2025-02-25:
Hello @eh2k after trying last night several hours to load samples as you mentioned ,i always have a error msg ..
i guess ll wait .. thanks for your time!