eh2k/squares-and-circles #119: Turing Machine engine crashes on load in the 2024-12-17 version

koswir, 2024-12-22:
Hey @eh2k 
In the latest version from 2024-12-17 I can not load the Turing Machine engine.
It happens on my 2 o_cs.
[image]
[image]

lessnerse, 2024-12-22:
Same to me on 3 devices…

 - i re-installed the Firmware 
 - I did format sd with and without keeping patches
 -  i swapped teens(ies)
 - I tried with loaded patches 
 - I tried with everything blanked on engines and outputs
 
[image]
[image]

eh2k, 2024-12-22:
@koswir,

Thanks for reporting. 
The TuringMachine should work again with version 8c3e8a9.

koswir, 2024-12-22:
> @koswir, @lessnerse
> 
> Thanks for reporting. The TuringMachine should work again with version [8c3e8a9](https://github.com/eh2k/squares-and-circles/commit/8c3e8a988b3cd4df76a7072984425f7081935c1b).

Thanks you! it works!

lessnerse, 2024-12-23:
Great! Thank you for the big support. 
as you see, your work impressed me for 3 uo_c‘s. :-)

closed 2025-01-11 by eh2k