eh2k/squares-and-circles #121: M-OSC/Waveforms Chords, Color parameter ....

eh2k, 2024-12-30:
>>I have removed the quantizer numbers - to avoid unnecessary confusion.

> Cmon please 😅 bring them back!
Only ppl with the perfect pitch can hear such complex and multi note chords (not me) so it was very useful...

_Originally posted by @koswir in https://github.com/eh2k/squares-and-circles/issues/117#issuecomment-2563189999_

koswir, 2025-01-01:
> > > I have removed the quantizer numbers - to avoid unnecessary confusion.
> 
> > Cmon please 😅 bring them back!
> > Only ppl with the perfect pitch can hear such complex and multi note chords (not me) so it was very useful...
> 
> _Originally posted by @koswir in [#117 (comment)](https://github.com/eh2k/squares-and-circles/issues/117#issuecomment-2563189999)_
> 
> Braids Renaissance Chords:
> 
> https://github.com/eh2k/squares-and-circles/blob/63799c4bd1c76ec005aa64357fe7274f0203d2d8/lib/braids/chords_stack.cc#L534-L585
> 
> Actually quite similar to the Plaits Chords
> 
> https://github.com/eh2k/squares-and-circles/blob/63799c4bd1c76ec005aa64357fe7274f0203d2d8/lib/plaits/dsp/chords/chord_bank.cc#L44-L90
> 
> In the current version, the colour parameter incrementation does not match the chords table. It may be possible to simply name the chords instead of ‘Colour’ as with the Plaits Chords.

@eh2k I do not get the first table.
How this is a 7th chord: {1, 6, 0, 0}, // 7th?
7th chord should contain first, third, fifth and seventh note in a scale.
If its + something (like the display you have removed) it should be +0, 2, 4, 6 scale degrees to a first note

eh2k, 2025-01-02:
@koswir 

Initially, I more or less simply copied the code from Renaissance-Repo without understanding the chord oscillators in detail. I had to make a few adjustments because there is an internal and an ‘engine-specific’ quantizer in S&C. The internal one is generally built into SC and affects all engines.

I have taken a closer look at the ‘Renaissance Chords’ code and must honestly say that the following points do not seem entirely logical to me:
  1. Only the first 8 chords from the table are used (only the top 3 bits from colour parameter are used for the chord table index)
     - The entries with index > 7 are actually unused code.
  3. Independent of the Color parameter value, two notes offsets 2 and 4 are added to the root note in all cases ( index << 11 ). Further note offsets are taken from the table `lib/braids/chords_stack.cc#L534`
     - As you also write, the values do not match the chords in the comments...
     - I wouldn't know how to name the chords either, as they still correspond to the quantizer steps.
  4. The logic of using the offsets to determine the frequencies from the quantiezer table and generating chords from them is also not clear to me (I think it's logical to use the mechanism with the turing machine, but I have no idea whether this makes any sense at all in terms of music theory).

I would prefer to use the same chords as in plaits chords (M-OSC/Chords), and extend it to 6 voices (Plaits only has a maximum of 4 voices). Then I would create the individual notes form the default semitone scale. 
  - Doesn't it make more sense to quantize each voice at the end than if there were 6x CV outputs individually for external oscillators?

eh2k, 2025-01-04:
@koswir 

In the latest update 4b23f97 there are the following changes to the chord oscillators:
  * If no quantiser is set, everything behaves as with WTx4. This means that the original Braids Chords table is used and the colour parameter interpolates between the values.
    [image]

  * If the SEMI Quantiser is set, the Plaits Chord table is used, where the default chords such as m7, m9, etc. are used (The value paramter becomes the selection as with M-OSC/Chords).
    [image]
  * For all other quantizer, the algorithm from Braids Renaissance is used, and the voice offsets are displayed as follows (by offsets I mean the index in the quantiser table):
    [image]

koswir, 2025-01-05:
@eh2k Great!
Thank you.
I am going to test it about monday.
👋

rootapprox, 2025-01-05:
> @koswir
>
> In the latest update [4b23f97](https://github.com/eh2k/squares-and-circles/commit/4b23f97a1614bb9d4cde06884e28d5affc4a76b0) there are the following changes to the chord oscillators:
>
> * If [...]

Looks and sounds dope @eh2k Thanks a lot for this update!

koswir, 2025-01-08:
> @koswir
>
> In the latest update [4b23f97](https://github.com/eh2k/squares-and-circles/commit/4b23f97a1614bb9d4cde06884e28d5affc4a76b0) there are the following changes to the chord oscillators:
>
> * If [...]

I love all the changes you've made - great work! I do have one suggestion, though:

Would it be possible to adjust the numbering of chord notes so they start from 1 instead of 0? 
In music theory, there's no such thing as a 'note zero.' For example, in a fifth, the notes are numbered 1 and 5, while a seventh chord consists of 1 (root), 3 (third), 5 (fifth), and 7 (seventh) - which is why it's called a 7th chord.

Currently, the numeration for a seventh chord goes like 0+2+4+6, but in my opinion it would be better to follow the correct sequence of 1+3+5+7. Similarly, the octave should be labeled as 8 (from 'octave'), not 7.

This change would make the numbering more intuitive and clear for a new user and aligned with standard musical theory.

---- 

There is still an issue with chord quantization when the transpose value is set to anything other than octaves in both transpose settings - the one in the I/O settings and the one on the main machine's panel.
When transpose is set to multiples of 12 (octaves), the chords quantize correctly. However, if the value is different, the chords behave unpredictably.

I'm not ready for digging into this behavior right now, though. 😅

eh2k, 2025-01-11:
@koswir @rootapprox 

thanks for your feedback.

@koswir - your suggestion (chord numeration) is now in the latest version aa5c787.

I'm closing this issue, if there is still something with the ‘general’ quantization, let's discuss it here. https://github.com/eh2k/squares-and-circles/issues/117

closed 2025-01-11 by eh2k